cesanta/mongoose #2527: `mg_http_serve_file()` fails to serve Android's `/proc` file's contents when file size is 0

smilediver, 2023-12-11:
We're using `mg_http_serve_dir()` to serve `/proc` dir on Android, but it fails to serve file's contents when the reported file size is 0, and just serves the file as if it is empty. This is probably an issue on all Linux based systems, since these files are special files and not actually empty even if they report size as 0.

The problem is in `mg_http_serve_file()`: it does `stat()` on the file, the size it gets is 0, and then it uses this size to report as content size and as a parameter on how many bytes `static_cb()` is supposed to send.

We're using older 7.6 mongoose version, but I checked the code in the repo and it seems the problem is still there.

scaprile, 2023-12-11:
When you open an issue, please honor the issue template, it is there to save yours and our time.
This is not a problem, Mongoose reads a file system and serves files.
If you want it to serve special files, please (besides properly filling the issue template) explain your use case so we can evaluate.

smilediver, 2023-12-12:
The use case is we need to expose those files to automation tools, so that application's and device's status can be inspected during testing.

scaprile, 2023-12-12:
Just for the sake of having fun, I mapped /proc to Apache and it does the same as we do.
So, we are not doing this.
Nothing stops you to mirror contents of interest and serve, or write your own virtual filesystem module.

closed 2023-12-12 by scaprile