cesanta/mongoose #2528: openssl

kashfi59, 2023-12-12:
1: add #include <openssl/bio.h> to src/tlc_openssl.h

2: in old openssl version, link error occur:
mongoose.c:(.text+0x6130): undefined reference to `BIO_get_data'
mongoose.c:(.text+0x8dfc): undefined reference to `BIO_get_new_index'
mongoose.c:(.text+0x8e08): undefined reference to `BIO_meth_new'
mongoose.c:(.text+0x8e14): undefined reference to `BIO_meth_set_write'
mongoose.c:(.text+0x8e20): undefined reference to `BIO_meth_set_read'
mongoose.c:(.text+0x8e2c): undefined reference to `BIO_meth_set_ctrl'
mongoose.c:(.text+0x8e40): undefined reference to `BIO_set_data'

kashfi59, 2023-12-12:
openssl version is 1.0.2

scaprile, 2023-12-12:
Please respect the guidelines and properly fill the issue template.
Your version is way too old
Mongoose is CI/CD tested in several environments.

closed 2023-12-12 by scaprile