cesanta/mongoose #2544: Unable to make a wss connection using 7.12 using qt5.15.16

hdabis1234, 2023-12-21:
I am unable to make a wss connection between qt 5.15.16 as client and mongoose 7.12 as a server. This works when using mongoose 7.11.

I managed to track down where this is breaking.

line 2007 in mongoose.c returns false
if (hm->method.len == 0 || hm->uri.len == 0) return -1;

I compared this to 7.11 and this line had moved.

More specifically when setting up the query in line 1999

Using the browser to connect I get hm->uri.ptr to be /?,,,,,

whilst when using qt I get hm->uri.ptr to be ?,,,,,

In qt this leads to hm->uri.len to equal 0 hence returning false.

- mongoose version: 7.12
- Compiler/IDE and SDK: visual studio 19_
- Target hardware/board: x86
- Target RTOS/OS (if applicable): Windows 10

scaprile, 2023-12-21:
Please do not go opening and closing issues,
Please search for other issues/questions, this has already been asked and answered before.
If none of those do not resolve your issue, then:
I don't understand what you say, please post a network capture file of the connection attempt and, as there is no built-in wss, I need to see what you are doing, it is _your_ responsibility to accept the connection, to start TLS handshake, and to upgrade to WebSocket. What *exactly* is it that is not working _for you_ in your scenario. There is no qt nor browser here, it is _your_ URL/URI requested in _your_ application, your client.

hdabis1234, 2023-12-21:
"Please do not go opening and closing issues," 
I pressed the button by mistake.

"Please search for other issues/questions, this has already been asked and answered before." 
Yes it was me who raised the issue.

"I don't understand what you say, please post a network capture file of the connection attempt and, as there is no built-in wss, I need to see what you are doing, it is your responsibility to accept the connection, to start TLS handshake, and to upgrade to WebSocket. What exactly is it that is not working for you in your scenario. There is no qt nor browser here, it is your URL/URI requested in your application, your client." 

I thought I was being clear by saying that the only variable I have is moving from mongoose 7.11 to 7.12. 
I was hoping that something obvious might be spotted by identifying where the issue is.

This is what I get in mg_http_parse(...)
s = "GET ?project-sessionid=lcvgdvwpefouaqacxfpr HTTP/1.1\r\nHost: 127.0.0.1:40857\r\nUpgrade: websocket\r\nConnection: Upgrade\r\nSec-WebSocket-Key: 8orZ42rQhEN7bWqi+drE0Q==\r\nSec-WebSocket-Version: 13\r\n\r\n"

I get the following events:
MG_EV_ACCEPT
MG_EV_TLS_HS
MG_EV_READ
MG_EV_ERROR
MG_EV_CLOSE

I do not get the MG_EV_HTTP_MSG event to upgrade the connection.

The source code is:

          if (ev == MG_EV_ACCEPT) {
		WebSocketServer *const webSocketServer = static_cast<WebSocketServer *>(conn->fn_data);
		struct mg_tls_opts tlsOpts;
		memset(&tlsOpts, 0, sizeof(tlsOpts));
		int rc;
		
		size_t size = 0;  // read CA certs from plain file
		char* data = mg_file_read(&mg_fs_posix, webSocketServer->sslCertFilePath_.c_str(), &size);
		tlsOpts.cert = mg_str_n(data, size);
		char* data2 = mg_file_read(&mg_fs_posix, webSocketServer->sslPrivateKeyPath_.c_str(), &size);
		tlsOpts.key = mg_str_n(data2, size);
		mg_tls_init(conn, &tlsOpts);
		free(data);
		free(data2);
	}
	else if (ev == MG_EV_HTTP_MSG) {
            ...
           mg_ws_upgrade(conn, mg_hm, NULL);	
       }

scaprile, 2023-12-21:
> "Please search for other issues/questions, this has already been asked and answered before."
 Yes it was me who raised the issue.

Then the answer is no other than the very same I gave you then https://github.com/cesanta/mongoose/issues/2432#issuecomment-1775055116

> I thought I was being clear by saying that the only variable I have is moving from mongoose 7.11 to 7.12. I was hoping that something obvious might be spotted by identifying where the issue is.

I beg to differ, there are lots of variables besides the one that spotted something for you in your scenario
You are not "identifying where the issue is", you are _spotting where your issue seems to be_, Without context, that means nothing. Please do as told in the former response, linked above. 
That what you show does not seem to be a proper URI.
Your connection is not being upgraded because no proper URI is being requested.
How can we do something about your URI that you send with your client ? OK, there is the possibility that we are not handling some requests properly. Let's check.
As traffic is TLS encrypted, repeat the procedure without TLS and capture what your client is requesting, so we can see if there is an actual problem in Mongoose URI parsing or (your client is just not playing fine or you are not calling it properly)
Mongoose 7.12 is less permissive than 7.11, if you read the release notes you'll see that we have tightened some checks for security's sake.
To capture, please use Wireshark or whatever you like, but attach a pcap or pcapng file, please do not paste screen snapshots.

hdabis1234, 2024-01-02:
So I managed to solve the issue but still not sure where the problem is.

This url works: wss://127.0.0.1:40857/?project-sessionid=lsovcxhnegimbdpydrtj
This url does not work: wss://127.0.0.1:40857?project-sessionid=lsovcxhnegimbdpydrtj

The second url is produced by Qt apis. I am not proficient in the spec of the url but the second form seems ok to me.

If you think that the issue is qt related then it can be closed here.

Thanks for your help.

scaprile, 2024-01-02:
As far as we are concerned, the second URL is missing the path component. The first one provides an empty path component.

closed 2024-01-02 by scaprile