cesanta/mongoose #2545: The meaning of "n <= 0" judgment in iolog function

xielingsen, 2023-12-23:
I'm using mongoose 7.12.
I encountered a problem that the connection will be closed after every request. After tracing, it was found that it was because mg_mgr_poll called read_conn, and read_conn did not obtain data after calling mg_io_recv. If iolog found n=-1, it would set c->is_closing = 1, causing the connection to close. I want to know the reason for doing this? Just because there is no data obtained when polling does not mean that the connection is intended to be closed, right?

cpq, 2023-12-23:
This is the code:

https://github.com/cesanta/mongoose/blob/06f823810789bf9cd04e6a7f198cc4897fa06160/src/sock.c#L244-L258

`recv()` reports error by returning negative value. If that's EAGAIN, we do not regard this is as an error. In other cases, we do: check `errno`, which gets printed by `read_conn()`.

closed 2023-12-23 by cpq

xielingsen, 2023-12-27:
> This is the code:
>
> https://github.com/cesanta/mongoose/blob/06f823810789bf9cd04e6a7f198cc4897fa06160/src/sock.c#L244-L258
>
> `recv()` reports error by returning negative value. If that's EAGAIN, we do [...]

I found that this issue was from mg_send.
Let's change the example "http-server" to:
`    if (mg_http_match_uri(hm, "/test")) {
      char buffer[256];
      mg_printf(c,
                "HTTP/1.1 200 OK \r\n"
                "Content-Type: application/octet-stream\r\n"
                "Content-Length: %u\r\n"
                "\r\n",
                256);

      mg_send(c, buffer, 256);
    } else {
`
Then the client can return successfully when accessing http://127.0.0.1/test for the first time, but it will definitely fail on the second access because the client continues to use the last connection and mongoose has closed the connection.

cpq, 2023-12-27:
@xielingsen please stop wasting yours and our time reporting false issues.