cesanta/mongoose #2548: mg_queue_prinf() sets wrong Content-Length / queue size.

T-a-n-d-E-m, 2023-12-29:
The following snippet sets the queue length to 2 (the length of the format string) not the expected 13 (the length of the final string generated).

// Create the queue and init it ...
char buffer[1024];
my_queue queue;
mg_queue_init(&queue, buffer, 1024);

// ... later on
const char* str = "Hello, World!";
mg_queue_printf(&queue, "%s", str);

scaprile, 2023-12-29:
There is an issue template for a reason: to save yours and our time. Please honor it.
Please follow the guidelines in the [documentation](https://mongoose.ws/documentation/),  [examples](https://github.com/cesanta/mongoose/tree/master/examples/) and [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples) available.

closed 2023-12-29 by scaprile

T-a-n-d-E-m, 2023-12-29:
Should I edit this existing issue to use the template or create a new issue?