cesanta/mongoose #2549: mg_queue_printf not setting correct queue size - what am I doing wrong?

T-a-n-d-E-m, 2023-12-29:
- My goal is: For the following program to return "Hello, World!":

```
#include "mongoose.h" // MG_VERSION "7.12" 
#include <pthread.h>

struct thread_data {
	struct mg_queue queue;
	int result;
};

static void start_thread(void *(*f)(void *), void *p) {
	pthread_t thread_id = (pthread_t) 0;
	pthread_attr_t attr;
	(void) pthread_attr_init(&attr);
	(void) pthread_attr_setdetachstate(&attr, PTHREAD_CREATE_DETACHED);
	pthread_create(&thread_id, &attr, f, p);
	pthread_attr_destroy(&attr);
}

static void *worker_thread(void* param) {
	struct thread_data *d = (struct thread_data *) param;

	static const size_t QUEUE_BUFFER_SIZE = 100;
	char buf[QUEUE_BUFFER_SIZE];
	mg_queue_init(&d->queue, buf, QUEUE_BUFFER_SIZE);

	d->result = 200;
	const char* str = "Hello, World!";
	mg_queue_printf(&d->queue, "%s", str); /* << USER ERROR HERE ???? */

	while (d->queue.tail != d->queue.head) usleep(1000);

	return NULL;
}

// HTTP request callback
static void http_request_callback(struct mg_connection *con, int event, void *event_data, void *usr_data) {
	(void) usr_data;

	if (event == MG_EV_HTTP_MSG) {
		struct mg_http_message *message = (struct mg_http_message *) event_data;
		struct thread_data *data = (struct thread_data *) calloc(1, sizeof(struct thread_data));
		start_thread(worker_thread, data);
		*(void **) con->data = data;
	} else if (event == MG_EV_POLL) {
		struct thread_data *data = *(struct thread_data **) con->data;
		size_t len;
		char *buf;
		if (data != NULL && (len = mg_queue_next(&data->queue, &buf)) > 0) {
			mg_http_reply(con, data->result, "Content-Type: text/plain\r\n", "%.*s\n", (int) len, buf); /* << USER ERROR HERE ???? */
			mg_queue_del(&data->queue, len);
			*(void **) con->data = NULL;
		}
	}
}

int main() {
	struct mg_mgr http_server;
	mg_mgr_init(&http_server);
	mg_log_set(MG_LL_DEBUG);
	struct mg_connection* connection = mg_http_listen(&http_server, "http://localhost:8000", http_request_callback, NULL);
	if(connection != NULL ) {
		for(;;) mg_mgr_poll(&http_server, 10);
	} else {
		fprintf(stderr, "mg_http_listen() failed\n");
	}
	mg_mgr_free(&http_server);
	return 0;
}
```

- My actions were: Compiled and execute the above test code:

>  % gcc main.c mongoose.c -o test -lpthread && ./test

... then make a request with curl, expecting Content-Length of 13 and "Hello, World!" as the text.

>  % curl -v localhost:8000
> *   Trying 127.0.0.1:8000...
> * Connected to localhost (127.0.0.1) port 8000 (#0)
> > GET / HTTP/1.1
> > Host: localhost:8000
> > User-Agent: curl/7.74.0
> > Accept: */*
> >
> * Mark bundle as not supporting multiuse
> < HTTP/1.1 200 OK
> < Content-Type: text/plain
> < Content-Length: 2
> <
> H
> * Connection #0 to host localhost left intact

Output from running ./test

> 2c5c9e49f 3 mongoose.c:4269:mg_listen   1 4 http://localhost:8000
> 2c5cb2676 3 mongoose.c:6719:accept_conn 2 5 accepted 127.0.0.1:48940 -> 127.0.0.1:8000
> 2c5cb2676 3 mongoose.c:6576:read_conn   2 5 snd 0/0 rcv 0/2048 n=78 err=0
> 2c5cb2680 3 mongoose.c:6587:write_conn  2 5 snd 76/2048 rcv 0/2048 n=76 err=0
> 2c5cb2681 3 mongoose.c:6576:read_conn   2 5 snd 0/2048 rcv 0/2048 n=-1 err=0
> 2c5cb2681 3 mongoose.c:4222:mg_close_co 2 5 closed

- My expectation was: I expected to see a Content-Length of 13 (strlen("Hello, World!")) and the text "Hello, World!"

- The result I saw: Content-Length of 2 (strlen("%s")) and H

- My question is: What am I doing wrong in the example program? Why is mg_queue_printf not setting the correct queue size?

# Environment
- mongoose version: 7.12

- Compiler/IDE and SDK: gcc 10.2.1, glibc 2.31

- Target hardware/board: 

> % cat /proc/cpuinfo
> processor       : 0
> vendor_id       : AuthenticAMD
> cpu family      : 6
> model           : 58
> model name      : Common KVM Processor v2
> stepping        : 9
> microcode       : 0x1000065
> cpu MHz         : 2850.000
> cache size      : 512 KB
> physical id     : 0
> siblings        : 1
> core id         : 0
> cpu cores       : 1
> apicid          : 0
> initial apicid  : 0
> fpu             : yes
> fpu_exception   : yes
> cpuid level     : 13
> wp              : yes
> flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 syscall nx rdtscp lm nopl cpuid tsc_known_freq pni pclmulqdq ssse3 cx16 sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx f16c rdrand hypervisor lahf_lm 3dnowprefetch vmmcall fsgsbase smep xsaveopt arat
> bugs            : fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
> bogomips        : 5700.00
> TLB size        : 1024 4K pages
> clflush size    : 64
> cache_alignment : 64
> address sizes   : 40 bits physical, 48 bits virtual

- Target RTOS/OS (if applicable): Debian 11

scaprile, 2023-12-29:
Please do not waste yours and our time opening several issues that are not Mongoose issues 
Duplicate of #2548
We will not check your code. You have working examples, work your way through them
https://github.com/cesanta/mongoose/tree/7.12/examples/multi-threaded
```
[multi-threaded]$ make
./example 
152897a3 3 mongoose.c:4269:mg_listen    1 4 http://localhost:8000
1528a42c 3 mongoose.c:6720:accept_conn  2 5 accepted 127.0.0.1:41632 -> 127.0.0.1:8000
1528a42c 3 mongoose.c:6578:read_conn    2 5 snd 0/0 rcv 0/2048 n=78 err=0
1528a437 3 mongoose.c:6589:write_conn   2 5 snd 66/2048 rcv 0/2048 n=66 err=0
1528a437 3 mongoose.c:6578:read_conn    2 5 snd 0/2048 rcv 0/2048 n=-1 err=0
1528a437 3 mongoose.c:4222:mg_close_con 2 5 closed
1528a437 2 main.c:50:worker_thread      done, cleaning up...
```
```
$ curl -v http://localhost:8000/
* Connected to localhost (127.0.0.1) port 8000 (#0)
> GET / HTTP/1.1
> Host: localhost:8000
> User-Agent: curl/7.61.1
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Length: 18         
< 
Send me POST data
* Connection #0 to host localhost left intact
$
```
If you see this very same issue with the example _as is_, then feel free to reopen this issue. Please do not open more issues.

Also, consider using the newer way in HEAD:
https://github.com/cesanta/mongoose/tree/master/examples/multi-threaded
https://mongoose.ws/documentation/tutorials/multi-threaded/

Please follow the guidelines in the [documentation](https://mongoose.ws/documentation/),  [examples](https://github.com/cesanta/mongoose/tree/master/examples/) and [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples) available.

closed 2023-12-29 by scaprile

T-a-n-d-E-m, 2023-12-30:
If the issue tracker isn't the place to ask, where does one go for help using Mongoose?