cesanta/mongoose #2569: RP2040 W5500 and DHCP

sas229, 2024-01-14:
- My goal is: Use the W5500 SPI Ethernet chip with the RP2040 and Mongoose via DHCP allocated IP
- My actions were: To build the example provided by Mongoose (https://mongoose.ws/documentation/tutorials/rp2040/pico-w5500/) and test with both static IP and DHCP allocated IP variants
- My expectation was: DHCP allocation would work
- The result I saw: 
For static IP allocation the device dashboard example works perfectly and I can log into the web app without issue from any device on my LAN:

[image]

However, for DHCP, the "link" comes up but an IP address is never allocated:

[image]

I considered whether this might be a router issue, so I used the same W5500 board with lwIP (and my own driver) and DHCP allocation works fine, so I don't think this is an issue with my LAN. As an aside the driver I used via lwIP uses DMA and so has much better throughput and also works via the Mongoose stack with a little adjustment of your example (which to be clear I am not testing here). For the lwIP based example, you can see that an IP address is properly set via DHCP, so I don't think there is a hardware problem on my part:

[image]

I have tried to use Wireshark to see if there is anything different about the DHCP packets sent via these two TCP/IP stacks, but I am out of my depth here and don't really know what this all means, but I thought it might be useful in debugging my issues. 

Wireshark DHCP packets for Mongoose:

Discover:
[image]
Request:
[image]

Wireshark DHCP packets for the lwIP version:

Discover:
[image]
Request:
[image]

All I can see is that lwIP sends extra options for maximum DHCP message size and requests the broadcast address, and the DHCP server identifier is 192.168.0.1 (my local gateway) via lwIP but 0.0.0.0 via Mongoose. All else looks the same. 

- My question is: Is there an issue with DHCP or am I doing something wrong? If I can provide anything else that might be useful, just let me know.

# Environment

- mongoose version: 7.12
- Compiler/IDE and SDK: GCC 11.2.1 arm-none-eabi and Pico SDK 1.5.1
- Target hardware/board: RP2040
- Connectivity chip/module: W5500 module
- Target RTOS/OS (if applicable): Bare metal (Pico SDK)

scaprile, 2024-01-14:
TL;DR
Please do not paste screen captures, I need to see a Wireshark capture to file to be able to guess what is going on in your network, all our examples have been tested and do work.
Please post a capture file of your device starting up and the corresponding Mongoose log dump

sas229, 2024-01-15:
Hopefully this is what you need. The router is a TPLink Deco M9.
[Wireshark capture files and Mongoose log dump.zip](https://github.com/cesanta/mongoose/files/13933527/Wireshark.capture.files.and.Mongoose.log.dump.zip)

scaprile, 2024-01-15:
You need to capture in a place where the traffic can be actually seen, and be kind not to include irrelevant traffic.
If your device MAC is 5a:38:83:2e:25:36, then use a capture filter for that.
You need suitable networking tools to do networking development, you need either a switch with monitor capability or a hub or a gateway that can run Wireshark or at least tcpdump, in order to be able to get the device traffic.
In your capture I see the device sends a DHCP Discovery, I don't see the answer because probably you captured in some random computer and some random switch in the middle of your network won't forward to that port any non-broadcast traffic that is not destined for that computer.
In your capture I see the device later sends a DHCP Request, I don't see the answer (same reason).
For some reason that I can't see (same reason), server id is not set in that request, and it should.
I can't see if your DHCP server never answers or our stack can't handle what it receives, all I see is that it stays in Req state, so probably your DHCP server never answers.

sas229, 2024-01-15:
Thanks for the advice. Unfortunately, it seems I don't have the right equipment.

Nevertheless, it seems odd that I can flash exactly the same hardware with firmware using an alternative DHCP stack, attached to the same network cable, switch, router etc., and the whole DHCP process works fine: on debug I can see the Discover, Offer, Request and Acknowledgement packets and the process is practically instant.

scaprile, 2024-01-15:
What is weird to me is that you are the only one having an issue like this, I offered my help to find the reason and if it turns out to be our problem, fix it. I can't do that if I don't see what is on the net. We could add extra debugging to make Mongoose log everything it sees, but that takes a lot of time.
Try to get that info and I will gladly work on it. Please reopen the issue when you have it. Regards.

sas229, 2024-01-16:
I appreciate the offer of help, I really do. I can also understand my problem seeming weird to you, for sure! I will see if I can find some proper network hardware to get the information requested.

Nonetheless, I noticed in a verbose debug log from rx_ip() that the packet received immediately after the DHCP request was sent contained my router IP address, but pkt->dhcp->siaddr which is passed to rx_dhcp_request_sel() was set to 0. It turns out that if I change line 4880 in rx_dhcp_client() from:

tx_dhcp_request_sel(ifp, ip, pkt->dhcp->siaddr); 

to:

tx_dhcp_request_sel(ifp, ip, pkt->ip->src);

then DHCP allocation works. Maybe my router is sending back a packet with information missing that you expected to be present? Or maybe what I have done above is inappropriate for reasons I don't understand?

scaprile, 2024-01-16:
Protocols should stay within their layer of incumbency, at least in theory.
A DHCP request takes the server IP address from that announced inside the DHCP datagram. If I understand you correctly, your DHCP server is not filling that.
The comment in that line is a reference to the section in the RFC named above, which reads:
>    If the parameters are acceptable, the client records the address of
>   the server that supplied the parameters from the 'server identifier'
>   field and sends that address in the 'server identifier' field of a
>   DHCPREQUEST broadcast message.

So, if that is correct, technically your gateway is not playing fair.
What we could do is a fallback like you did, something like:
```
  // select IP, (4.4.1) (fallback to IP source addr on foul play)
  tx_dhcp_request_sel(ifp, ip, pkt->dhcp->siaddr ? pkt->dhcp->siaddr : pkt->ip->src);
```

sas229, 2024-01-16:
Thanks for the explanation. It has helped my understanding a great deal.

cpq, 2024-01-16:
> tx_dhcp_request_sel(ifp, ip, pkt->ip->src);
> 
> then DHCP allocation works. Maybe my router is sending back a packet with information missing that you expected to be present? Or maybe what I have done above is inappropriate for reasons I don't understand?

This is perhaps the issue, thank you Sam!

Sergio, your fallback logic seems good, could you please send a PR

scaprile, 2024-01-16:
Sure, I was waiting for the OP to test it as all our DHCP servers play nice.

sas229, 2024-01-16:
I have tested it just now and it works perfectly. Thanks!

cpq, 2024-01-16:
@sas229 Sam, your efforts are much appreciated!

scaprile, 2024-01-16:
@cpq Should we add a minimum functionality tcpdump-like to be enabled on cases like this ? Only dump our frames and those for us, but do it in a way that can later be opened with Wireshark. Easily enabled with an allegoric macro like MG_TCPDUMP. Except for old and bald salty dogs, no one is expected to have a hub and getting device traffic becomes hard for most people.

PR #2571 (2024-01-16): Fallback to IP source address on DHCP servers not filling sid
Closes #2569