cesanta/mongoose #2570: Http client POST intermittently fails.

jimellert, 2024-01-14:
- My goal is:  To retrieve data from an https server (using openssl).
- My actions were:  Send a POST to https://httpbin.org/post,  20 times in a row, and verify response.
- My expectation was:  Each post responds correctly within 1 sec.
- The result I saw:    2 posts respond correctly, then 1 post times-out, then 1 post is correct, then 2 fail, etc.
- My question is:   Am I doing something wrong, or can this be fixed?
Public reproducible build:     https://github.com/jimellert/mongooseDemo1.git
Run test number 3 to reproduce the problem.

# Environment

- mongoose version:  7.12
- Compiler/IDE and SDK:   Visual Studio Community 2022 ver 17.8.3,   Visual C++
- Target hardware/board:   Dell Precision 7730,   Raspberry Pi 4 Model B Rev 1.4
- Connectivity chip/module: 
- Target RTOS/OS (if applicable):   Windows 10,   Ubuntu 22.04.3 LTS

jimellert, 2024-01-14:
```
//****************************************************************************
// mongooseDemo1
// James W. Ellert
//****************************************************************************
#include <iostream>
#include <iomanip>
#include <string>
#include "mongoose.h"

using namespace std;
static constexpr const char* HDR_CTFORM = "Content-Type: application/x-www-form-urlencoded\r\n";
static mg_mgr g_mgMgr;

struct Tx {
	string  m_method;
	string  m_url;
	string  m_headers;
	string  m_body;
	string* m_responseBuf;
	int32_t m_respCode;
};
static Tx g_tx;

static void
mongooseClientCallback (mg_connection* nc, int ev, void* evData) {
	switch (ev) {
	case MG_EV_READ: {
		cout << "mongooseClientCallback::MG_EV_READ" << endl;
		//string buf ((char*)nc->recv.buf, nc->recv.len);
		//cout << buf << endl;
		} break;

	case MG_EV_CONNECT: {
		cout << "mongooseClientCallback::MG_EV_CONNECT" << endl;
		auto tx = (Tx*)nc->fn_data;
		mg_tls_opts opts = { 0 };
		mg_tls_init (nc, &opts);
		mg_str host = mg_url_host (tx->m_url.c_str());
		string uri  = mg_url_uri  (tx->m_url.c_str());
		string msg;
		msg = tx->m_method + " ";
		msg += uri + " HTTP/1.1\r\n";
		msg += "Host: " + string(host.ptr, host.len) + "\r\n";
		msg += "Content-Length: " + to_string(tx->m_body.length()) + "\r\n";
		msg += tx->m_headers + "\r\n";
		msg += tx->m_body + "\r\n";
		msg += "\r\n";

		mg_send (nc, msg.c_str(), msg.length());
		} break;

	case MG_EV_HTTP_MSG: {
		cout << "mongooseClientCallback::MG_EV_HTTP_MSG" << endl;
		auto tx = (Tx*)nc->fn_data;
		auto hm = (mg_http_message*)evData;
		tx->m_responseBuf->append (hm->body.ptr, hm->body.len);
		tx->m_respCode = mg_http_status(hm);
		} break;
	}
}

static int32_t
post (const string& url, const string& headers, const string& body, string* response) {
	g_tx.m_method      = "POST";
	g_tx.m_url         = url;
	g_tx.m_headers     = headers;
	g_tx.m_body        = body;
	g_tx.m_responseBuf = response;
	g_tx.m_respCode    = 0;
	mg_http_connect (&g_mgMgr, url.c_str(), mongooseClientCallback, &g_tx);

	for (;;) {
		mg_mgr_poll (&g_mgMgr, 200);
		if (g_tx.m_respCode != 0)
			return g_tx.m_respCode;
	}

	return g_tx.m_respCode;
}

static int32_t
get (const string& url, const string& headers, string* response) {
	g_tx.m_method      = "GET";
	g_tx.m_url         = url;
	g_tx.m_headers     = headers;
	g_tx.m_responseBuf = response;
	g_tx.m_respCode    = 0;
	mg_http_connect (&g_mgMgr, url.c_str(), mongooseClientCallback, &g_tx);

	for (;;) {
		mg_mgr_poll (&g_mgMgr, 200);
		if (g_tx.m_respCode != 0)
			return g_tx.m_respCode;
	}

	return g_tx.m_respCode;
}

static void
testPost() {
	string body = "Hello from mongooseDemo1.";
	string response;
	auto respCode = post ("https://httpbin.org/post", HDR_CTFORM, body, &response);
	if (respCode != 200 || response.find(body) == string::npos)
		cout << "testPost failed." << endl;
	else
		cout << "testPost successful." << endl;

	//cout << response << endl;
}

static void
testBigGet() {
	string response;
	auto respCode = get ("https://httpbin.org/bytes/8000", "", &response);
	if (respCode != 200 || response.length() < 8000)
		cout << "testBigGet failed." << endl;
	else
		cout << "testBigGet successful." << endl;

	//cout << std::hex;
	//for (int i = 0; i < response.length(); ++i)
	//	cout << std::setw(2) << std::setfill('0') << (int)response.at(i);
}

static void
testManyPosts() {
	for (int32_t i = 0; i < 20; ++i)
		testPost();
}

static void
printCmds() {
	cout << "Options:" << endl;
	cout << "         1  :  testPost."      << endl;
	cout << "         2  :  testBigGet."    << endl;
	cout << "         3  :  testManyPosts." << endl;
}

int main (int argc, char** argv) {
	if (argc < 2) {
		printCmds();
		return (0);
	}

	mg_mgr_init(&g_mgMgr);

	int32_t cmd = atoi(argv[1]);

	switch (cmd) {
	case 1:  testPost      (); break;
	case 2:  testBigGet    (); break;
	case 3:  testManyPosts (); break;
	default: printCmds     (); return (0);
	}

	mg_mgr_free(&g_mgMgr);

	cout << endl << "====Test complete====" << endl;
}
```

scaprile, 2024-01-14:
I'm sorry, I'm too lazy to check your code. Even though, I see that it doesn't even resemble what we show in our documentation
Please follow the guidelines in the [documentation](https://mongoose.ws/documentation/),  [examples](https://github.com/cesanta/mongoose/tree/master/examples/) and [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples) available.
There is an http client example, use that to test what you are trying to do:
https://mongoose.ws/documentation/tutorials/http-client/
If after that you still have problems:
- Please sniff your network (using Wireshark, for example) and observe the traffic.
If, after your network inspection, you find an actual issue, feel free to reopen this issue and provide us with pertinent information (including useful network captures) in order to reproduce and fix the issue. We don't inspect your code for errors, please send the minimum piece of code that can reproduce the issue

jimellert, 2024-01-21:
As requested, the code has been simplified to the minimum necessary to reproduce the problem, and switched to C and to a style that matches the http-client tutorial.  I'm still working on what I can get out of a Wireshark capture (since this is a secure connection).  Please do not waste any of your time checking my code.  As before, the github repo is:  https://github.com/jimellert/mongooseDemo1.git

Reminder, the problem is that the body of the requests are not always returned when MG_EV_HTTP_MSG is called.  (Note, above the previous code was blocking and timing-out and the latest code does not.  So no more talk of timeouts.)  Printing the receive buffer in MG_EV_READ shows sometimes the body has not been read when we get MG_EV_HTTP_MSG.

jimellert, 2024-01-21:
```
//****************************************************************************
// mongooseDemo1
// James W. Ellert
//****************************************************************************
#include <string.h>
#include <stdio.h>
#include "mongoose.h"

static const char*   s_url  = "https://httpbin.org/post";
static const char*   s_body = "Hello from mongooseDemo:";
static struct mg_mgr g_mgMgr;

struct Msg {
	int  m_tryNum;
	char m_response[1024];
	int  m_respCode;
};

static void
fn (struct mg_connection* nc, int ev, void* ev_data) {
	struct Msg* msg = (struct Msg*)nc->fn_data;

	switch (ev) {
	case MG_EV_READ: {
		printf ("-fn::MG_EV_READ\n");
		//printf ("%s\n", (char*)nc->recv.buf);
		} break;

	case MG_EV_CONNECT: {
		printf ("\n-fn::MG_EV_CONNECT\n");
		struct mg_tls_opts opts = { 0 };
		mg_tls_init (nc, &opts);

		struct mg_str host    = mg_url_host (s_url);
		int           bodyLen = (int)strlen(s_body) + ((msg->m_tryNum < 10) ? 1 : 2);

		mg_printf (nc, "POST %s HTTP/1.0\r\n"
			"Host: %.*s\r\n"
			"Content-Length: %d\r\n"
			"Content-Type: application/x-www-form-urlencoded\r\n\r\n"
			"%s%d\r\n"
			"\r\n",
			mg_url_uri(s_url), (int)host.len, host.ptr, bodyLen, s_body, msg->m_tryNum);
		} break;

	case MG_EV_HTTP_MSG: {
		printf ("-fn::MG_EV_HTTP_MSG\n");
		struct mg_http_message* hm = (struct mg_http_message*)ev_data;
		strncpy_s (msg->m_response, sizeof(msg->m_response), hm->body.ptr, hm->body.len);
		size_t i = (hm->body.len >= sizeof(msg->m_response)) ? sizeof(msg->m_response)-1 : hm->body.len;
		msg->m_response[i] = 0;
		msg->m_respCode = mg_http_status(hm);
		} break;

	case MG_EV_ERROR:
		printf ("-fn:MG_EV_ERROR: %s\n", (char*)ev_data);
		break;
	}
}

int main (int argc, char** argv) {
	mg_mgr_init (&g_mgMgr);

	for (int32_t i = 0; i < 20; ++i) {
		struct Msg msg;
		msg.m_tryNum   = i;
		msg.m_respCode = 0;

		mg_http_connect (&g_mgMgr, s_url, fn, (void*)&msg);

		for (;;) {
			mg_mgr_poll (&g_mgMgr, 200);
			if (msg.m_respCode != 0)
				break;
		}

		if (msg.m_respCode != 200 || !strstr (msg.m_response, s_body)) {
			printf ("-----Test Post failed:\n");
			printf ("respCode: %d\n", msg.m_respCode);
			printf ("response: \"%s\"\n", msg.m_response);
		} else {
			printf ("-----Test Post successful.\n");
		}
	}

	mg_mgr_free (&g_mgMgr);
	printf ("\n-----Test complete.\n");
}
```

scaprile, 2024-01-22:
I've just run 2000 iterations of your program with no TLS. Not a single failure.
However, with TLS, I get 0-length messages. Though it works with 7.12 release. You should try that until we sort this out.

@cpq this is weird, I simplified this to a minimum. Works with no TLS. Works with TLS in 7.12. Fails (not always) with TLS in HEAD. Both OpenSSL and MbedTLS tried. The OP uses OpenSSL.

[pp.zip](https://github.com/cesanta/mongoose/files/14003247/pp.zip)

7.12: steady as follows, no failures
```
-fn::MG_EV_CONNECT
d25b213 2 pp.c:27:fn                    EV_HTTP_MSG: 200, 388

-fn::MG_EV_CONNECT
d25b4a1 2 pp.c:27:fn                    EV_HTTP_MSG: 200, 388
```
HEAD: 
```
-fn::MG_EV_CONNECT
d303264 2 pp.c:27:fn                    EV_HTTP_MSG: 200, 388

-fn::MG_EV_CONNECT
d3034f4 2 pp.c:27:fn                    EV_HTTP_MSG: 200, 388

-fn::MG_EV_CONNECT
d30378c 2 pp.c:27:fn                    EV_HTTP_MSG: 200, 0
```
The third attempt fails. We seem to always send the same amount of data; however the server response in the third case is smaller
[postls.zip](https://github.com/cesanta/mongoose/files/14003288/postls.zip)
If you devise a way to see what is going on without having the server's private key... otherwise we should mimic the server and I'm afraid we won't see failures... as all our tests pass...

jimellert, 2024-01-22:
Wireshark apparently has the ability to decrypt TLS traffic by supplying it with a SSLKeyLogFile.  I will look into this tomorrow.

jimellert, 2024-01-22:
Status update:   I'm not getting far.  Choices I've seen are:
Add the SSL_CTX_set_keylog_callback to mongoose to write a SSLKEYLOGFILE,  or,
Run a local httpbin using https://github.com/postmanlabs/httpbin.git.
These are both beyond my capability.  Sorry guys, I wish I could be more help.

scaprile, 2024-01-23:
Don't worry, I didn't check for a local httpbin, that might help. I'll see what I can do. I'm not friends with OpenSSL but somehow we'll see what is going on in there.

cpq, 2024-01-23:
The issue that you observe is inside Mongoose.
Mongoose reads raw encrypted data into the `c->rtls` buffer. Then, it feeds that data to the TLS library.
Currently, if server finished sending the data, it closes the connection. Mongoose reads data into `c->rtls`, so `c->rtls` has some data to be decrypted, then Mongoose reads `0` from the socket, indicating that server has closed the connection.
And Mongoose closed the connection too, despite it has some data in `c->rtls` yet to be decrypted!

The PR I've sent, fixes that issue and decrypts everything before closing the connection.

scaprile, 2024-01-23:
Great, yesterday night I managed to decrypt the traffic and came to the conclusion that decryption was sometimes failing and delivering 0-length messages.
Just tested off that branch in #2587 and it works OK

BTW, to get OpenSSL to SSLKEYLOGFILE, in tls_openssl.c:
```
static void cb(const SSL *ssl, const char *line) {
    FILE *fp;
    fp = fopen("keylog.txt", "a");
    if (fp == NULL) MG_ERROR(("Failed to create log file"));
    fprintf(fp, "%s\n", line);
    fclose(fp);
}

  if (!s_initialised) {
    SSL_library_init();
//    s_initialised++;                    // <-- COMMENT OUT
  }
  MG_DEBUG(("%lu Setting TLS", c->id));
  tls->ctx = c->is_client ? SSL_CTX_new(SSLv23_client_method())
                          : SSL_CTX_new(SSLv23_server_method());
  if ((tls->ssl = SSL_new(tls->ctx)) == NULL) {
    mg_error(c, "SSL_new");
    goto fail;
  }
  SSL_set_session_id_context(tls->ssl, (const uint8_t *) id,
                             (unsigned) strlen(id));
  if (!s_initialised) {                 // <-- ADD AND CLAIM "initialised" HERE
    SSL_CTX_set_keylog_callback(tls->ctx, cb);
    s_initialised++;
  }
 ```

jimellert, 2024-01-23:
Good find Sergey!  And good find Sergio - capturing that SSLKEYLOGFILE patch could be useful in the future if you have to debug encrypted traffic.  Thanks you guys!   I'll watch for the merge into the master.

commit 8dc62c25 (2024-01-23): Merge pull request #2587 from cesanta/tls

Fix #2570: close only when TLS buffers drained
commit d8fec13d (2024-01-23): Fix #2570: close only when TLS buffers drained
closed 2024-01-23 by scaprile