cesanta/mongoose #2574: Unbounded growth in transmit buffer web WebSockets

EndlessDex, 2024-01-17:
## Environment

- mongoose version: 7.12
- Compiler/IDE and SDK: gcc version 11.4.0
- Target hardware/board: Linux 5.15.0-91-generic x86_64

## Setup

We are running the Mongoose web server to provide HTTP and WebSocket service for an embedded device. For this testing, we replicated it on a generic Ubuntu server to ensure easy reproducibility.

Our client application uses WebSockets to subscribe to a data-stream from our backend. After the initial subscription, the backend is expected to continuously serve updates until mongoose detects a disconnect (MG_EV_CLOSE). 

For a minimal reproducible example, we set mongoose to wake on timer and send data over the active WebSocket connections (In our actual implementation, the timer callback calls a select on a localhost socket to see if information is available and should be forwarded over the WebSocket). This removes the connection from mongoose to our backend but functionally serves the same purpose.

## Bug Symptom

The following figure is a graph of memory (RAM) usage over time for various processes from 2023-01-08T19:30Z to 2023-01-09T01:30Z. A period of roughly 6 hours. The Y-Axis is in MegaBytes. The blue process (PID 786467) is mongoose. The memory useage of mongoose will continue to grow until all available memory is consumed.
[image]

## Bug Cause

Through our own debugging we have nailed down the specific cause of the issue.

Microsoft Edge (and other browsers) have a feature to "Put inactive tabs to sleep" automatically after a period of non-interaction. Edge specifically enables this by default whereas Google Chrome has this disabled by default.

This sleep state seems to have an odd effect on the WebSocket connection. It does not send a disconnect but also does not continue processing the WebSocket. From the server side, the system TCP buffer begins to fill up as soon as the browser tab goes to sleep. We are not sure but we presume this to mean that the client is no longer ACKing packets and that the server is attempting to automatically buffer and retransmit them as TCP is designed to do.

For a period of time, there is no visible issue. The server TCP buffer is filling but mongoose does not know (or care) about this. It continues to do its job in transmitting data; however, eventually the TCP buffer is completely filled. Mongoose DOES recognize this (via `mg_iotest` and either EPOLL or some OS-dependant variation) and sets the `is_writeable` flag to False. This tells mongoose to stop calling `mg_io_send` (which wraps the OS `send`).

This is where a very useful feature of mongoose becomes a curse. Typically when a high-level send call is made (i.e. `mg_ws_send`) mongoose will `calloc` a buffer for all data to be sent and then drain some or all of the buffer each `mg_poll` cycle. This allows sending arbitrarily large amounts of data to the (relatively) small OS TCP buffer. We  use this feature ourselves to enable the sending of 300kB blobs of data without needing to manually split it into TCP-buffer-sized chunks.

With the TCP buffer filled and not draining, mongoose's internal buffering takes over and begins growing in size. Each new call to `mg_ws_send` will cause the internal buffering to allocate more and more memory. This growth is completely unbounded. As long as you keep calling `mg_ws_send`, it will keep growing.

From what we have seen, this issue NEVER resolves itself. We have a logfile of a system that was in this state for over 10 days before it ran out of RAM and crashed.

If the browser page is reopened, the connection will drain normally and all used memory will be freed by the garbage collector.

If the browser page is closed, the connection will be dropped and all used memory will be freed by the garbage collector

## Reproducing

[This repo](https://github.com/EndlessDex/mongoose_issue_debug) contains a minimal example to reproduce the issue.

There are three real files:

1. mongoose_main.cpp: source file for the webserver
1. nonsense.h: a garbage string to send over the WebSocket
1. index.html: basic webpage that will open a WebSocket and append all received message to `<body>`

The other files are compilation related. If you have cmake, the example can be built by running `make mongoose` in the root directory of the repo. `make run` will start the mongoose server. `make all` will do both.

On the browser side, using Microsoft Edge you can set the amount of time before a tab goes to sleep in the settings. The timout can be set as low as 30 seconds.

[image]

With the example running, open the webpage with Edge. You should see text begin to appear once the WebSocket connection is established. Now minimize the browser. This will cause the tab to go to sleep after a short time (though not exactly the time specified in the settings).

Each call to `WebSocketSend` will be logged as well as the mongoose TX and RX buffer sizes. Initially the `write_conn` calls are also visible though they go away after the TCP buffer fills and `is_writeable` becomes permanently false. Once this happens the TX buffer can be seen to increase linearly for the rest of the runtime.

A logfile of this is attached. [logfile.txt](https://github.com/cesanta/mongoose/files/13968037/logfile.txt)

- See line 831 for the beginning of buffer growth.
- See line 1382 for when the browser was reopened.
- Note that 1,845,248 bytes had accumulated at this time.

scaprile, 2024-01-17:
I don't really think I have to clone your repo and spend my time browsing your code. Sorry about that.
Why don't you please trim this down to the simplest minimum piece of code reproducing your issue and I'll gladly check it, and if it is our fault, fix it.
Please reopen this issue when you've done that

EndlessDex, 2024-01-17:
```
    mg_timer_add(&mgr, WEBSOCKET_READ_INTERVAL, MG_TIMER_REPEAT, WebSocketSend, &mgr);
```

```
static void WebSocketSend(void *arg) {
    static unsigned char message[BUFSIZE] = {0};
    static char message_enc[BUFSIZE] = {0};

    int len = snprintf(reinterpret_cast<char *>(message), BUFSIZE, "%d: %s", packet_counter++, nonsense);

    // Encode the data
    len = mg_base64_encode(message, len, message_enc, BUFSIZE);

    for (auto it = websocket_map.begin(); it != websocket_map.end(); ++it) {
        mg_ws_send(it->first, message_enc, len, WEBSOCKET_OP_TEXT);
        MG_DEBUG(("WebSocketSend: %s", it->second.c_str()));
    }

    struct mg_mgr *mgr = (struct mg_mgr *)arg;
    struct mg_connection *c;
    for (c = mgr->conns; c != NULL; c = c->next) {
        if (c->is_websocket) {
            MG_DEBUG(("%p %lu : txBuf %p/%u/%u/%u    rxBuf %p/%u/%u/%u", c, c->id,
                      c->send.buf, c->send.len, c->send.size, c->send.align,
                      c->recv.buf, c->recv.len, c->recv.size, c->recv.align));
        }
    }
}```

EndlessDex, 2024-01-17:
As I state in my original comment, the repo is already in the minimal state. Please take the time to actually read the full content of the issue before closing it next time.

scaprile, 2024-01-17:
Sorry, I fail to see how that gets to run. I can't compile that.
I've already been to your repo and had to navigate to find what is supposed to be some file that actually might have some reference to how you might perhaps be calling mongoose library functions. In that file I had to scroll several pages. We seem to differ in the definition of "minimal".
If you are a paying customer or have a support contract, please contact support@cesanta.com and we'll gladly serve you, otherwise, please be kind to us and submit something small and simple that we can build and run to reproduce your issue.

EndlessDex, 2024-01-17:
The issue has been sent to the support team. I AM a paying customer and look forward to the response.

First you ask for less code and then you ask for more. Both times without addressing the mentioned issue at all. I still do not believe you have actually read my post.

The source you need is in the linked repo at the mentioned filepath. It it a waste of MY time to continue discussing this with you when you have put in seemingly zero effort.

scaprile, 2024-01-17:
I did read your post, I did visit your repo. I need to see how you call our functions, and your last post does not show that.
Do you call  mg_timer_add() only once ?
How do you call the event manager Data is not sent when you call mg_ws_send(), it is sent when you call the manager.
Are you keeping everything in a single thread ?
Quite likely you already know that, but since you did not identify yourself, I need to ask.
Our support team will see your message tomorrow, I don't run Edge and I can't build a small app to try what you say. Unfortunately, we'll have to wait until tomorrow. I'll reopen and pin this issue for you.

cpq, 2024-01-18:
@EndlessDex  thank you!

Mongoose uses poll/select/epoll to figure out, can it drain the `c->send` output IO buffer to the peer, or not. Apparently when a browser stalls network IO, the TCP/IP stack on Mongoose side fills its socket buffer, and poll/select/epoll reports that this connection cannot drain, thus `c->send` keep the data and grows as you write more to it. The connection does not close, it just stalls.

It looks like that the question here is - how can we detect this situation? I believe it should be handled in the following way:
1. We can record the time of the last successful write. The easiest way is to store it in the `c->data` scratch buffer. Something like this:
```c
struct conndata {
  uint64_t last_write_ms;  // The timestamp of the last write
};
...
void ev_handler(struct mg_connection *c, int ev, void *ev_data) {
  struct conndata *cd = (struct conndata *) c->data;
  if (ev == MG_EV_WRITE) {
    cd->last_write_ms = mg_millis(); // Record last IO time
  }
  ...
```
2. Then, the `WebSocketSend()` function can respect two metric parameters: that last io time, and the size of the output buffer. If either of the two is larger than a certain threshold that you can tune, don't write to the connection:
```c
  uint64_t now = mg_millis();
  for (auto it = websocket_map.begin(); it != websocket_map.end(); ++it) {
    struct mg_connection *c = it->first;
    struct conndata *cd = (struct conndata *) c->data;
    if (now - cd->last_write_ms > TOO_LONG || c->send.len > TOO_BIG) {
      MG_INFO(("Connection %lu is stalled, not sending anything to it"));
      continue;
    }
    ...
```

Would that be a solution to the problem?

A nitpick: Mongoose's printf functions support `%m` non-standard modifiers that can base64-encode. You can create your own custom printer that e.g. prints your structures as binary or JSON. Your snippet may look like this:

```c
static void WebSocketSend(void *arg) {
    char buf[BUFSIZE];
    size_t len = mg_snprintf(buf, sizeof(buf), "%d: %M", packet_counter++, mg_print_base64, 5, "hello");
    // Send buf to all WS clients
```

Or even without an intermediate buffer - base64 encode directly to the `c->send` IO buffer:
```c
static void WebSocketSend(void *arg) {
    for (auto it = websocket_map.begin(); it != websocket_map.end(); ++it) {
        mg_ws_printf(it->first, WEBSOCKET_OP_TEXT, "%d: %M", packet_counter, mg_print_base64, 5, "hello");
        MG_DEBUG(("WebSocketSend: %s", it->second.c_str()));
    }
   packet_counter++;
```

cpq, 2024-02-11:
Closing this. Feel free to reopen if there is more data on the issue.

closed 2024-02-11 by cpq