cesanta/mongoose #2589: Mongoose documentation SSL/TLS tutorial is inaccurate for mg_tls_opts.

jimellert, 2024-01-25:
- My goal is:  To understand how to initialize TLS on the server side.
- My actions were:  Follow the example in the SSL/TLS tutorial.
- My expectation was:   Correct operation.
- The result I saw:   OpenSSL complaint about "not enough data" for certs.

The mg_tls_opts .cert and .key fields are supposed to be assigned the actual cert and key strings, not a path to the cert and key files.
The "How to load credentials" section correctly shows using mg_file_read to load the cert and key files and assign the cert and key strings to mg_tls_opts.
But the "TLS for servers", TLS for clients", and "Two-way TLS" sections show using mg_unpacked to assign a path to the .ca, .cert, & .key variables.  These sections should be updated to use mg_file_read unstead of mg_unpacked.

scaprile, 2024-01-26:
Thank you for your opinions.  I beg to differ. There is nothing incorrect nor outdated in that tutorial.
The very same section you mention, in the following paragraphs, shows other two forms in which you can load your credentials. 
> They can also be loaded from a file in an embedded filesystem, using mg_unpacked() as shown above:
>
>struct mg_tls_opts opts = {.ca = mg_unpacked("/certs/ca.pem")};
>
>This accesses the file directly from the code, where it has been embedded.

"above" in the first sentence, makes reference to what you've seen in the server and client sections.

> Finally, certificates and keys can be defined as C strings in source code:

That can be seen in the http-restful-server example, linked in the server section, the device-dashboard example, and probably others too.

These three ways are also tested every day on every push:
https://github.com/cesanta/mongoose/blob/dd6458262d0869bbd72d8ca0aea52dcbe9643b33/test/unit_test.c#L1286-L1289
https://github.com/cesanta/mongoose/blob/dd6458262d0869bbd72d8ca0aea52dcbe9643b33/test/unit_test.c#L786-L787
https://github.com/cesanta/mongoose/blob/dd6458262d0869bbd72d8ca0aea52dcbe9643b33/test/unit_test.c#L790-L794

Mongoose is embedded oriented, and while we serve the OS world, we aim to support devices with no built-in filesystem.
The client and server sections do not say "do this to load your credentials", those sections are explaining that you have to initalize TLS at that point, and how real world examples taken from actual examples in our repo, linked in the final lines of that section, do it:
TCP server:
https://github.com/cesanta/mongoose/blob/dd6458262d0869bbd72d8ca0aea52dcbe9643b33/examples/tcp/main.c#L60-L63
HTTP client:
https://github.com/cesanta/mongoose/blob/dd6458262d0869bbd72d8ca0aea52dcbe9643b33/examples/http-client/main.c#L31-L34
Both show you how to build the embedded file system, and the TCP server example does not carry the packed filesystem itself so it will generate it on the fly for you. It also shows both client and server.

´´´
[tcp]$ make CFLAGS_EXTRA="-DMG_TLS=MG_TLS_OPENSSL -lssl -lcrypto"
cc ../../test/pack.c -o ./pack
./pack                     certs/ss_server.pem certs/ss_ca.pem certs/ss_client.pem > packed_fs.c
cc main.c mongoose.c packed_fs.c        -W -Wall -Wextra -g -I.   -DMG_ENABLE_PACKED_FS=1 -DMG_TLS=MG_TLS_OPENSSL -lssl -lcrypto -o example                     
./example
21064ce 2 main.c:56:sfn                 SERVER is listening
2106532 2 main.c:19:cfn                 CLIENT has been initialized
2106532 2 main.c:83:timer_fn            CLIENT connecting
2106532 2 main.c:21:cfn                 CLIENT connected
2106532 2 main.c:58:sfn                 SERVER accepted a connection
21078c3 2 main.c:43:cfn                 CLIENT sent data
21078c3 2 main.c:67:sfn                 SERVER got data: Hi, there
21078c3 2 main.c:31:cfn                 CLIENT got data: Hi, there
2108ac3 2 main.c:34:cfn                 CLIENT disconnected
2108ac3 2 main.c:71:sfn                 SERVER disconnected
´´´

closed 2024-01-26 by scaprile

jimellert, 2024-01-26:
I'm sorry.  I did not specify MG_ENABLE_PACKED_FS when compiling.  I think that's what caused mg_unpacked(path) to not work for me.  My bad.