cesanta/mongoose #2614: Simplify cmsispack mongoose_custom.h

cpq, 2024-02-19:
This is not good for a customer:

https://github.com/cesanta/mongoose/blob/054eca8a3eae4465a328756fcf28c98749f88fc9/extra/cmsis-pack/Mongoose/config/mongoose_custom.h#L60-L79 

It should look like plain, simple and direct mongoose_custom.h

Please do the following:
1. Simplify this header
2. Add TLS builtin option
3. Rename this `extra` to `support` and add to the master branch

scaprile, 2024-02-19:
This has been done on current examples and is scheduled to be ported to our next package version. The pack version will have it just like the regular example version in the repo. That will occur some time after our next release. We always release new packs when we release a new version 
(In my defense: that format was intended to be used inside the wizard, the user was not expected to see that, just twiddle the options and be able to run Mongoose over his preferred RTOS and stack)

- "extra" has been agreed to be named "extra", "support" may be confusing (tech support ? what support ? support for what ?). Any other names ?
- This has also been kept somehow hidden not to confuse our users, as there are lots of repeated, slightly different. examples and vendor files around, things that only make sense inside a pack but are necessary to build it. Users getting pieces from this tree will find nothing works, because this stuff only works with a pack installed. When more vendors add their pack variant, this may grow worse.

cpq, 2024-02-19:
Thanks Sergio!

Well, users ARE supposed to see and edit that file.
I am going to use it in my workshop in EW with ARM. So if it is possible, I'd make a release of cmsis pack earlier, maybe with  a minor release number, we might have few iterations before we stabilise.

`support` sounds better to me than`extra`, but let's brainstorm the best name. We probably need to move some stuff from test/ there, e.g. pack.c - does it really belong to test?

scaprile, 2024-02-19:
Let's switch to dev chat

closed 2024-02-28 by cpq