cesanta/mongoose #2617: page404 unexcepted Content-Type

TwinIsland, 2024-02-20:
- **My goal is**: Enable the server to respond with a custom 404 page specified in `opts` when a request is made to an unknown resource.
- **My actions were**: I configured opts with `page404` to a custom HTML page. I then attempted to access a non-existent resource via http://localhost:1146/unknownstuff.
- **My expectation was**: I expected the server to respond with the custom 404 page, including setting the `Content-Type` header to `text/html`.
- **The result I saw**: The `Content-Type` header of the response is determined by the extension of the requested URI, not the MIME type of the custom 404 page itself.
- **My question is**: Shouldn't the Content-Type for the 404 response be consistent and reflect that the resource is an HTML page, especially given that the option is named page404? Would it be advisable to enforce a Content-Type of text/html for the 404 page to ensure proper rendering by browsers?

https://github.com/cesanta/mongoose/blob/93148e322340a1ce8d9a1eec2d2983698ad72b8f/mongoose.c#L2788C1-L2792C4

# Environment

- mongoose version: [7.12](https://github.com/cesanta/mongoose/releases/tag/7.12)
- Compiler/IDE and SDK: gcc 11.4.0/vscode
- Target hardware/board: x64
- Connectivity chip/module: i5 12th 12600k
- Target RTOS/OS (if applicable): null

scaprile, 2024-02-20:
I think the order of 2790 and 2791 has to be reversed, and you'll get what you want (and probably what the dev intended, there is no point in serving image/png when a requested image fails, for example)

TwinIsland, 2024-02-20:
> I think the order of 2790 and 2791 has to be reversed, and you'll get what you want (and probably what the dev intended, there is no point in serving image/png when a requested image fails, for example) Can you please test ?

Yea it works as expected, thanks!

closed 2024-02-23 by cpq