cesanta/mongoose #2619: Reading larger POST data with enabled tls fails

jcorporation, 2024-02-20:
- My goal is: Read post requests from a https socket issued by javascript
- My actions were: read the data in `MG_EV_HTTP_MSG` with `struct mg_http_message *hm = (struct mg_http_message *) ev_data;` as used in your examples
- My expectation was: mongoose reads the complete post data
- The result I saw: 
 
Mongoose does not read the complete post data from the socket. I checked with wireshark and the browser sends the complete message.

The problem occurs only with enabled ssl (openssl in my case)

The issue is reproduceable with your http-restful-server example. Attached are the mongoose logfile with hexdumping enabled, the tcpdump and the keylog file. Code is the same as in #2603 but only with a larger body.

[test.zip](https://github.com/cesanta/mongoose/files/14350876/test.zip)

# Environment

- mongoose version: latest master
- Compiler/IDE and SDK: `gcc version 13.2.0 (Ubuntu 13.2.0-4ubuntu3)`
- Target hardware/board: `Linux t14 6.5.0-15-generic #15-Ubuntu SMP PREEMPT_DYNAMIC Tue Jan  9 17:03:36 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux`

scaprile, 2024-02-21:
@cpq I see Content-Lenght is present, I see POST data is apparently complete, it is larger than an Ethernet frame but since this is the loopback it has been transferred on a single frame. I see Mongoose buffer is 2048, content-length is smaller, frame size is larger. I see Mongoose logs 2048, looks like it stops collecting after that, once it needs to enlarge its buffer.

@jcorporation Can you try with a client on a different host ? If that is no burden to you, I'd like to see how it works on regular wire.

jcorporation, 2024-02-21:
Same test with a iPhone as client.

[test-wlan.zip](https://github.com/cesanta/mongoose/files/14363720/test-wlan.zip)

jcorporation, 2024-03-03:
Any news on this?

cpq, 2024-03-03:
@jcorporation not yet
looking at it.

commit 01e7cd5f (2024-03-14): Merge pull request #2646 from cesanta/tls

Fix #2619 - use mg_tls_pending() API to check for buffered TLS data
commit 3ac9bd04 (2024-03-14): Fix #2619 - use mg_tls_pending() API to check for buffered TLS data