cesanta/mongoose #2640: Issue with 'ENCRYPT' in 'mongoose.h'

gvanem, 2024-03-05:
On Windows and `MG_VERSION "7.13"`, this little C-program (called  `mongoose-ntddndis-error.c`):
```c
#include <mongoose.h>
#include <iphlpapi.h>
int main(void)
{
  return (0);
}
```

fails to compile. From a `clang-cl -c -MD -I./externals mongoose-ntddndis-error.c`:
```c
In file included from mongoose-ntddndis-error.c:2:
In file included from f:\gv\WinKit\Include\10.0.22621.0\um\iphlpapi.h:1447:
In file included from f:\gv\WinKit\Include\10.0.22621.0\shared\netioapi.h:152:
f:\gv\WinKit\Include\10.0.22621.0\shared\ntddndis.h(2137,5): error: expected identifier
 2137 |     ENCRYPT
      |     ^
./externals\mongoose.h(1223,17): note: expanded from macro 'ENCRYPT'
 1223 | #define ENCRYPT 1  // specify whether we're encrypting
      |                 ^
1 error generated.
```
Since `ntddndis.h` uses `ENCRYPT` in an `enum`. This fixes it:
```c
#include <mongoose.h>
#undef ENCRYPT
#include <iphlpapi.h>
...
```

But I feel Mongoose should not "*pollute*" the namespace with `ENCRYPT`  etc.
So can `ENCRYPT` (and `DECRYPT`) please be renamed to `MG_ENCRYPT` or something?

scaprile, 2024-03-05:
@cpq This is in the PD `tls_aes128.[ch]` files.
How should we proceed, do we prepend `MG_` in both files or just `#undef` at the header file header ?

cpq, 2024-03-06:
@scaprile prepending `MG_` is the way to go, we should not be polluting user's namespace

gvanem, 2024-03-06:
Here is another case; but only a redefinition warning. For this program:
```c
#include <string.h>
#include <malloc.h>
#include <mimalloc/mimalloc-override.h> // From https://github.com/microsoft/mimalloc
#include <mongoose.h>
int main (void)
{
  return (0);
}
```

I get:
```c
In file included from mongoose-ntddndis-error.c:6:
./externals\mongoose.h(499,9): warning: 'realpath' macro redefined [-Wmacro-redefined]
  499 | #define realpath(a, b) _fullpath((b), (a), MG_PATH_MAX)
      |         ^
./externals\mimalloc/mimalloc-override.h(28,9): note: previous definition is here
   28 | #define realpath(f,n)           mi_realpath(f,n)
      |         ^
1 warning generated.
```

Since `_fullpath()` allocates memory internally, *mimalloc* has this override function [`mi_realpath()`](https://github.com/microsoft/mimalloc/blob/dev/src/alloc.c#L900C42-L900C50) for it.

cpq, 2024-03-06:
Thank you @gvanem !

scaprile, 2024-03-06:
@cpq How should we proceed with this last one ? Redefinition will depend on the order of inclusion, if we `undef` then the other one will warn. There is no undef in mimalloc-override https://github.com/microsoft/mimalloc/blob/master/include/mimalloc-override.h . Checking for `mi_realpath` and using theirs may not be a good choice.

cpq, 2024-03-06:
I think the best way to deal with the realpath() type of issues, as well as all the other issues that arise from using things like mimalloc and similar tools, is to avoid using `arch_win32.h` - basically, our headers.

@gvanem  So the realpath issue is different from the ENCRYPT macro, and should be fixed this way:
1. Set `-DMG_ARCH=DMG_ARCH_CUSTOM`
2. Define your own "mongoose_custom.h" and include whatever deemed necessary - mimalloc, whatever.

@scaprile so please do just the ENCRYPT fix and don't touch realpath

closed 2024-03-06 by scaprile