cesanta/mongoose #2649: How to  upload a file by https?(device-dashboard)

moneypi, 2024-03-15:
- My goal is: upload file by https
- My actions were: 
I modify code as below
[image]
and change directory to `examples/device-dashboard` and execute `make`
then I can open url `https://127.0.0.1:8443/#/update`
- My expectation was:
I wish I can upload file by `Upload new firmware .bin file`
[image]

- The result I saw: If the size of upload file is larger than 1G, it seems like I can't finish uploading.
- My question is:
What should I do to upload a file by https?

# Environment

- mongoose version: 7.13
- Compiler/IDE and SDK: gcc 9.4.0
- Target hardware/board: x86/black sesame A1000L
- Connectivity chip/module: 
- Target RTOS/OS (if applicable): Linux

scaprile, 2024-03-15:
Please follow the guidelines in the [documentation](https://mongoose.ws/documentation/),  [examples](https://github.com/cesanta/mongoose/tree/master/examples/) and [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples) available.

https://mongoose.ws/documentation/tutorials/file-uploads/
https://mongoose.ws/documentation/tutorials/tls/
https://github.com/cesanta/mongoose/tree/master/examples/file-transfer

closed 2024-03-15 by scaprile

moneypi, 2024-03-15:
I'm sorry, and I have to admit that I'm not familar with ssl or tls.
I rebuild the `device-dashboard` with command `make CFLAGS_EXTRA="-DMG_TLS=MG_TLS_MBED -lmbedtls -lmbedcrypto -lmbedx509"`, and ,in fact, I can upload small file, like 1k, in `https://127.0.0.1:8443/#/update` but I can't upload large file, no error, no crash.
And I can upload large file by `http://0.0.0.0:8000/#/update`.

scaprile, 2024-03-15:
Your problem is not TLS, this is not and end product, please read the docs and examples and do accordingly. The device dashboard is an example and the firmware upload feature is an in-development feature designed to download firmware and we have no embedded module with gigs of flash so we won't transfer nor test for gigs of data. It is not a one-size-fits-all file transfer tool.
You are expected to write your code based on Mongoose, reading our tutorials, following our examples, knowing your system, knowing your tools, and fitting to your needs.