cesanta/mongoose #2656: Facing issue in accessing FLASH_SIZE register on the STM32H7

Simayon, 2024-03-22:
- My goal is: To perform OTA update on STM32H745-NUCLEO
- My actions were: I ran the example code for STM32H743 with Dashboard UI
- My issue is: Precise data access violation (PRECISERR)
- My question is: Is there any mechanism preventing me from accessing the address directly

# Environment

- mongoose version: 7.12
- Compiler/IDE and SDK: STM32 CubeIDE version 1.14.1
- Target hardware/board: STM32H7

Accessing the below given address value results in an error,
[image]

Called from the function mg_flash_size() given below,
[image]

I am providing the full diagnostic below 
[image]

scaprile, 2024-03-22:
STM32H743 is STM32H743, belongs to one family, has one reference manual.
STM32H745 is STM32H745, belongs to another family, has another difference reference manual. The fact that their arbitrary numbering differs by 2 is meaningless. Those can be as different as any other of the dozens of the different models.
The example says H74**3** for a reason: it supports the H74**3** and has been tested for the H74**3**
If you want to use any other of the dozens of different models STM32 offers, whether their arbitrary numbering differs by 2, by 20 or by 200, you have to download its reference manual, read it, see the differences, write the code for it. It might be the same, it might be not. If there is no example for it, we did not test it.

Regarding your scenario, our example is not for Cube, it is for baremetal. We know what we setup, we test it, we don' enable things we do not need, we do not let other's libraries to do anything out of our control. Cube is Cube.

WRT your specific question, we don't know what Cube or you have set or not set. Add to that what was explained above, and that address might not even exist in your processor.

closed 2024-03-22 by scaprile

Simayon, 2024-03-22:
Thanks scaprile for the explanation, that makes sense. I did look at the reference manual for the STM32H745 and the FLASH_SIZE_REG address does appear to be the same, with a value of 0x400 (1024) corresponding to the 2MB flash.

Just to confirm, there shouldn't be any specific read permissions required to access this register address, correct?  

In the meantime, I've gone ahead and hardcoded 1024 in place of MG_REG(FLASH_SIZE_REG) which seems to have resolved the issue, but I'm still unsure if this is the best approach. 

[image]

scaprile, 2024-03-22:
We use that address to support all members of the family, that have different sizes.
We do not enable MPUs nor Trustzone, but your Cube code might be doing it. Are you using an RTOS ? That can also be restricting hw access. We don't.

Simayon, 2024-03-22:
That explanation really helps!  Now I understand why things were working the way they were.  I'm using an RTOS, so that might be causing the access issue.  In the meantime, I might just stick with the hardcoded value for now.