cesanta/mongoose #2660: Is it save to call free() after passing data to mg_ws_send()?

nook24, 2024-03-23:
- My goal is:
I'm working on a shared library, that gets loaded into a single threaded C program. My goal is to use the WebSocket server of mongoose to push data to external clients.
Due to the main program is single threaded, I can not block the main thread as it would slowdown everything else.

- My actions were:
I have started a new thread where the webserver runs.
The main program calls callback functions inside of my library. I encode the passed data into JSON and handover the JSON-String to `mg_ws_send()`. After that, I immediately free the JSON object.

This works without any issues so far, but I just wanted to make sure that this is the right way to do this. 
My code looks like this (only necessary parts):
```C
// Web server thread function
void *web_server_thread(void *data)
{
	// Initialize Mongoose event manager
	mg_mgr_init(&mgr);
	mg_log_set(ws_debug_level);
	mg_http_listen(&mgr, ws_listening_address, fn, NULL); // Create listener

	mg_wakeup_init(&mgr); // Initialise wakeup socket pair
	while (!ws_exit_flag)
	{
		mg_mgr_poll(&mgr, 1000);
	}
	mg_mgr_free(&mgr);

	return NULL;
}

// Function to stop the web server thread
void stop_web_server_thread()
{
	ws_exit_flag = true;
}

// asynchronous push method that will not block the main app
static void push_to_clients(struct mg_mgr *mgr, const char *buf)
{
	struct mg_connection *c;
	for (c = mgr->conns; c != NULL; c = c->next)
	{
		mg_ws_send(c, buf, strlen(buf), WEBSOCKET_OP_TEXT);
	}
}

// Callback that gets called by the main program
static int cb_host_check_data(int cb, void *data)
{
	json_object *hostcheck_object;
	const char *json_string;

	nebstruct_host_check_data *ds = (nebstruct_host_check_data *)data;

	// Encode the current host check event as JSON object
	hostcheck_object = nebstruct_encode_host_check_as_json(ds);

	// Convert the JSON object to a string
	json_string = json_object_to_json_string(hostcheck_object);

	// Push host check to web sockets clients (async)
	push_to_clients(&mgr, json_string);

	// Release resources
	json_object_put(hostcheck_object);

	return 0;
}

// Broker initialize function
int nebmodule_init(int flags, char *args, nebmodule *handle)
{
	// Start the web server in a separate thread
	if (pthread_create(&tid, NULL, web_server_thread, NULL) != 0)
	{
		return NEB_ERROR;
	}

	// Register callbacks
	neb_register_callback(NEBCALLBACK_HOST_CHECK_DATA, nebapi_module_handle, 0, cb_host_check_data);
}

int nebmodule_deinit(int flags, int reason)
{
	stop_web_server_thread();
	pthread_join(tid, NULL);
}
```

This is the complete code: https://github.com/nook24/neb-api/blob/main/src/nebapi.c

Thanks in advance. Also awesome library!

# Environment

- mongoose version: _7.13_
- Compiler/IDE and SDK: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
- Target hardware/board: amd64 and arm64
- Connectivity chip/module: _?_
- Target RTOS/OS (if applicable): Ubuntu and RHEL Linux

scaprile, 2024-03-23:
Hello Daniel,
we don't check user code, can't afford the time to do that. At birdseye view, you don't seem to be doing what we suggest. Here is a tutorial on what we suggest for multithreading: https://mongoose.ws/documentation/tutorials/multi-threaded/ , and the following includes your "one to many" (12m) scenario: https://github.com/cesanta/mongoose/tree/master/examples/multi-threaded-12m
Have you already checked that ?
Please do it and come back with any issues you may find.

nook24, 2024-03-24:
Hi Sergio,
thanks for the tutorial, this was relay helpful. I will replace my calls to `mg_ws_send` with `mg_wakeup` and only use `mg_ws_send` inside of the HTTP request handler as recommended by the documentation and examples.

In the 12m scenario I honestly don't understand why the worker thread gets started, inside of the HTTP request handler
https://github.com/cesanta/mongoose/blob/d9a6e85b65957674ae4eff375a5a05d195d35975/examples/multi-threaded-12m/main.c#L45-L53

If i run the example code the worker thread immediately starts even without an active client connection at all. So I'm not sure why the worker thread gets started inside of the HTTP request handler in this case.

The `multi-threaded` example on the other hand is clear, it starts a long running task for each new client.

scaprile, 2024-03-24:
Well, the worker thread is not just a "worker" thread per se... 
We need to start that task at some time. It is supposed to be a task that does something and the clients are interested in getting that something, so that task execution is independent of the existence of clients. At least in this example, as I see it.
In the example, we start it as soon as Mongoose starts. You can do it the way you want, to fit your purpose; just keep track that you've already started it, as requests may come from several connections at the same time and you'll have a lot of joy watching race conditions happen.

scaprile, 2024-03-24:
Closing this, you can keep asking here anyway

closed 2024-03-24 by scaprile

nook24, 2024-03-25:
Thanks for your feedback and time, I appreciate this :)