cesanta/mongoose #2679: support for openssl-3

Necktwi, 2024-04-03:
- My goal is: host https website
- My actions were: i've given ssl certificate path in `mg_tls_opts` struct
- My expectation was: https protocol is served without an issue
- The result I saw: 
```
C8C725967F000000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../openssl-3.0.13/crypto/encode_decode/decoder_lib.c:101:No supported data to decode. Input type: PEM
C8C725967F000000:error:0A080009:SSL routines:SSL_use_PrivateKey_file:PEM lib:../openssl-3.0.13/ssl/ssl_rsa.c:182:
1a8e976c 1 mongoose.cpp:404:mg_error    11 0xc Invalid SSL key, err 1
C8C725967F000000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../openssl-3.0.13/crypto/encode_decode/decoder_lib.c:101:No supported data to decode. Input type: PEM
C8C725967F000000:error:0A080009:SSL routines:SSL_use_PrivateKey_file:PEM lib:../openssl-3.0.13/ssl/ssl_rsa.c:182:
1a9106cf 1 mongoose.cpp:404:mg_error    12 0xc Invalid SSL key, err 1
```
- My question is: I found that issue is because of openssl-3 as per https://github.com/openssl/openssl/issues/16696. is it true? if support is already provided, is there an example that works with openssl-3?

# Environment

- mongoose version: 655f5db996d2210394d98ded3339796b40e5a2b4
- Compiler/IDE and SDK: ZZZ version _X.Y_
- Target hardware/board: Core3566
- Connectivity chip/module: rk3566
- Target RTOS/OS (if applicable): Gentoo (should not matter)

scaprile, 2024-04-03:
We don't provide C++ examples, you provide your certificates, you write your code, you load them; we are unable to guess what you do.
All our examples do work, they are run on each library change.
You are likely not providing the right certs/keys with proper formatting.
Please follow the guidelines in the [documentation](https://mongoose.ws/documentation/),  [examples](https://github.com/cesanta/mongoose/tree/master/examples/) and [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples) available.
https://mongoose.ws/documentation/tutorials/tls/#how-to-load-credentials
PEM files must start with dashes

closed 2024-04-03 by scaprile

Necktwi, 2024-04-03:
oh! how is my same mongoose server is working fine when built against openssl-1! may be some glitch in the matrix ?\_(?)_/?

scaprile, 2024-04-03:
We run unit tests on GH "ubuntu-latest", that is Ubuntu 22.04, that uses openssl 3.02
The following builds and runs the http-restful-server example off this repo on Ubuntu 22.04

[openssl.zip](https://github.com/cesanta/mongoose/files/14858672/openssl.zip)

tested it works.

Necktwi, 2024-04-04:
Thank you for confirmation! I'll look into the issue with my software stack.