cesanta/mongoose #2680: Handshake with TLS 1.3 server fails using MbedTLS

markjfisher, 2024-04-03:
- My goal is: To get latest mongoose client working with servers that use TLS 1.3
- My actions were: To use mongoose http client to establish connections with server using https
- My expectation was: it would connect and negotiate TLS handshake correctly
- The result I saw: Added logs etc below - the handshake returns -0x7780
- My question is: How can I get the latest version working like the older version I upgraded from (7.4)

# Environment

- mongoose version: 7.11
- Compiler/IDE and SDK: cmake targetting either gcc or clang
- Target hardware/board: x86 replicated on both linux and msys2
- Connectivity chip/module: n/a
- Target RTOS/OS (if applicable): n/a

I've attached 2 logs of sessions with 2 different servers, httpbin.org (called tls-httpbin-ok.txt), and a google cloud service (called tls-5cs-bad.txt).

The 2 endpoints I'm using are:
- https://httpbin.org/get (good)
- https://5card.carr-designs.com/state?table=blue\&player=THOMC (bad)

The logs include full mbed tls logging during the TLS negotiation.
In the google (5cs) case, the logs error on 3rd pass through MBEDTLS_SSL_SERVER_HELLO

The log shows:

```
client state: MBEDTLS_SSL_SERVER_HELLO
=> parse server hello
=> read record
=> fetch input
in_left: 0, nb_want: 5
in_left: 0, nb_want: 5
ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
<= fetch input
dumping 'input record header' (5 bytes)
0000:  15 03 01 00 02                                   .....
input record: msgtype = 21, version = [0x301], msglen = 2
=> fetch input
in_left: 5, nb_want: 7
in_left: 5, nb_want: 7
ssl->f_recv(_timeout)() returned 2 (-0xfffffffe)
<= fetch input
dumping 'input record from network' (7 bytes)
0000:  15 03 01 00 02 02 28                             ......(
got an alert message, type: [2:40]
is a fatal alert message (msg 40)
mbedtls_ssl_handle_message_type() returned -30592 (-0x7780)
mbedtls_ssl_read_record() returned -30592 (-0x7780)
<= handshake
23ea8c33 1 mongoose.c:1363:mg_error     1 676 TLS handshake: -0x7780
```

In the working case, the lines differ at the input record msgtype log, there, it says:
```
input record: msgtype = 22, version = [0x303], msglen = 95
```

0x303 being TLS 1.2, I'm unsure why in the error case it is reporting 0x301, as all indications from RFCs I've read suggest this is TLS1.0.

I noticed when doing a connection to this endpoint with curl that it negotiates purely on TLS 1.3, but the httpbin.org drops back to 1.2.

google hosted service curl logs:
```
* Connected to 5card.carr-designs.com (142.250.179.243) port 443
* ALPN: curl offers h2,http/1.1
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
*  CAfile: C:/msys64/clang64/etc/ssl/certs/ca-bundle.crt
*  CApath: none
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384 / x25519 / RSASSA-PSS
```

httpbin curl logs:
```
* Connected to httpbin.org (54.147.29.229) port 443
* ALPN: curl offers h2,http/1.1
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
*  CAfile: C:/msys64/clang64/etc/ssl/certs/ca-bundle.crt
*  CApath: none
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES128-GCM-SHA256 / secp256r1 / rsaEncryption
```

In all cases in my client code, I'm using mbed tls version that does not support 1.3, which is shown in both logs as "The SSL configuration is tls12 only."

If I roll the mongoose client back to 7.4, which we have been using in our project up to this point, then the handshake does not error.
I have not, however, been able to reproduce the mbed tls logging with that client as I did with 7.11, despite setting logging flags and log levels the same as when using the latest client code.

We are attempting to contact the person who is hosting the 5cs google service to try and get them to get google to downgrade the TLS version to see if this is the actual error, as I am a bit lost in all the logs, but 1.3 is the only difference I can see between the 2 endpoints.

If there is any additional information I can give to help this problem, please let me know.

[tls-5cs-bad.txt](https://github.com/cesanta/mongoose/files/14853127/tls-5cs-bad.txt)
[tls-httpbin-ok.txt](https://github.com/cesanta/mongoose/files/14853128/tls-httpbin-ok.txt)

scaprile, 2024-04-03:
Hello Mark,
we just _use_ MbedTLS, there is nothing special in that besides possible API differences.
Our tests run on Windows, Mac, and Ubuntu 22.04 (what Github provides) and they all pass.
You should start by using 7.13, and you could even try our built-in TLS1.3
However, a bit of googling yields: "_The error you are getting means that you have received a fatal alert from the server.
This fatal alert was sent after the server received the ClientHello message from your client.
This means that the server couldn’t find common parameters for a TLS handshake. Usually it is the server can’t support the proposed ciphersuites, however it could be other cases such as unsupported eliptic curves and hashes_.", so you should check your credentials and what those servers expect from you, AND, in the specific code you are using, if there is an mbed configuration file, that will have specifics on what it will support, so you have to make it match, or remove that file and expect your OS will solve that.

markjfisher, 2024-04-03:
Hi! And thanks for the update.

This problem is happening on my and other users machines over different OSes, the only thing we have done is upgrade the mongoose client code, which is why I brought it here.

I had seen it's reporting a fatal error, 21 response being a MSG_ALERT, 22 being MSG_HANDSHAKE, and that 40 represents ERR_SSL_VERSION_OR_CIPHER_MISMATCH. The logs show the earlier negotiation of ciphers is identical between the 2 endpoints I have logged, showing all 104 ciphers are the same.

I'm really trying to narrow down why only upgrading mongoose client causes this issue, downgrading fixes it. No mbed tls changes (as we're just using the system installed libs), just mongoose.c/h file changes in our code.

Our code is simply trying to request JSON served over https, there's no auth/credentials involved here. It's as simple as can be.

I'll continue to dig into this, I'm trying to reproduce with a local nginx as a reverse proxy and controlling tls versions.

scaprile, 2024-04-03:
Hmm... that's weird.
There's been a lot of re-write in the TLS interface to make room for our own stack, and we've been favoring embedded-friendly features (elliptic curves and lighter-weight algorithms). I'm not familiar with pre-7.6.
So you have a CA cert and you just validate it, but there is a cypher exchange anyway and that is failing... Make sure your cert file contents start with dashes, and check how you pass that cert in options. It should fail in a different way if that was the case, but, it doesn't harm to try.

markjfisher, 2024-04-03:
Also apologies, we are using 7.13, not 7.11 as I reported, I'll update the post

scaprile, 2024-04-03:
I just tried your Google endpoint in my Linux with both OpenSSL and MbedTLS and it works with no hassle. Looks like this is a problem with your specific version of MbedTLS.
We've been very reluctant to rely on OS libraries, for a while we've been building against specific versions compiled on-the-fly. I think a quick fix for your issue can be to do that. In examples/http-client:
```
$ make mbedtls
$ make CFLAGS_EXTRA="-DMG_TLS=MG_TLS_MBED mbedtls/library/libmbedtls.a mbedtls/library/libmbedcrypto.a mbedtls/library/libmbedx509.a"
```

markjfisher, 2024-04-03:
Thanks for the update. I've just fixed it in our code.
Very sorry for the noise. It turns out we were not reading the certificate correctly with the upgrade in clients, yet that was somehow working with some sites but not others.

Many thanks for your feedback on this and sorry to have wasted your time.

closed 2024-04-03 by markjfisher

CaptainTeemo, 2024-04-18:
@markjfisher hello Mark,
I just got the same issue, could you please share the details about how you fixed?

scaprile, 2024-04-18:
@markjfisher :
> Very sorry for the noise. It turns out we were not reading the certificate correctly with the upgrade in clients, yet that was somehow working with some sites but not others.

This is not a Mongoose issue, please do not do necroposting not thread hijacking, 
If there you have an actual issue, please open an issue, _honor the issue template_ so we can know your scenario and reproduce the issue if necessary, and let us know _exactly_ what you are doing
Please follow the guidelines in the [documentation](https://mongoose.ws/documentation/),  [examples](https://github.com/cesanta/mongoose/tree/master/examples/) and [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples) available.