cesanta/mongoose #2686: S390 test gets stuck after WS with tls_builtin

scaprile, 2024-04-04:
S390 test gets stuck after WS with tls_builtin:
```
4d758  2 unit_test.c:2746:w2            Got WS, 8192
4d758  2 unit_test.c:2755:w2            8192
4d758  3 sock.c:305:write_conn          3 6 snd 10/2048 rcv 0/10240 n=10 err=0
4d758  3 sock.c:305:write_conn          2 5 snd 18/2048 rcv 8/2048 n=18 err=0
4d758  3 sock.c:295:read_conn           3 6 2:0:0 18 err 0
4d758  3 sock.c:295:read_conn           2 5 0:8:0 10 err 0
4d758  3 ws.c:187:mg_ws_cb              WS PONG
4d759  3 ws.c:199:mg_ws_cb              2 WS CLOSE
4d759  3 sock.c:305:write_conn          3 6 snd 4/2048 rcv 0/10240 n=4 err=0
4d759  3 sock.c:305:write_conn          2 5 snd 15/2048 rcv 0/2048 n=15 err=0
4d759  3 net.c:151:mg_close_conn        2 5 closed
4d759  1 event.c:31:mg_error            3 6 socket error
4d759  3 net.c:151:mg_close_conn        3 6 closed
4d765  3 net.c:151:mg_close_conn        1 4 closed
4d765  3 net.c:249:mg_mgr_free          All connections closed
```
and there it stays forever

cpq, 2024-05-16:
@scaprile Could you test this one if it is fixed already please?

scaprile, 2024-05-16:
It's been failing consistently every night, waiting for us to find the fix.
Latest merge adding WolfSSL removed this test.
I will shortly add it back.

scaprile, 2024-05-16:
Interestingly, it seems to be working now...

closed 2024-05-16 by scaprile