cesanta/mongoose #2689: QoS1 and QoS2 cannot be implemented with the API

RicoBiedermann, 2024-04-05:
**- My goal is:** 
Implement QoS 1 and QoS 2 for MQTT 3.1.1 and MQTT 5 based on the example: https://mongoose.ws/documentation/tutorials/mqtt-client/

**- My actions were:** 
I have readed the mqtt specs and some sites from mqtt broker companys. I have also readed the mongoose code.
https://www.emqx.com/en/blog/introduction-to-mqtt-qos (this site shows it realy good)
http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718099
https://docs.oasis-open.org/mqtt/mqtt/v5.0/os/mqtt-v5.0-os.html#_Toc3901026
https://www.hivemq.com/blog/mqtt-essentials-part-4-mqtt-publish-subscribe-unsubscribe/
http://www.steves-internet-guide.com/understanding-mqtt-qos-levels-part-1/

**- My expectation was:** 
That I'm able to implement QoS 1 and QoS 2 with the tutorial.

**- The result I saw:**
I think it is not possible to archiv this with the current mongoose mqtt api.
To archiv this, we need a way to do a "retransmition" where I can set the packet-id and the duplication flag. 
If I sent a message with QoS1 and I dont get a MQTT_PUBACK for that message, I need to send it again.  But i need to send it with the same packet id and the duplication flag needs to be true. A Broker or a other client has no other way of recognising that it is a message that has already been sent, or that several messages are the same. 
If the message has a new packet id, it is a new message. 
If I do a subscription, I also need this information when I receive a message to determine if it is a new message or not.
On QoS2 i need the option to send a MQTT_CMD_PUBREL.

**- My question is:** 
Is there a way that I don't know yet?
Could it be a way to expand the mg_mqtt_message struct with a field for the retainflag, duplicationflag and to add the user porperties like the mg_mqtt_opts struct? 
The mg_mqtt_message struct could be used for mg_mqtt_pub() and mg_mqtt_sub() and with some changes in this functions it could work. 

# Environment

- mongoose version: 7.13
- Target hardware/board: imx8mp
- Target RTOS/OS (if applicable): Debian based linux

RicoBiedermann, 2024-04-05:
Why do you have closed that issue? 

I have checked the documentation, examples and the tutorials. And also I have checked now the unit_test. 
None of this provides an answer to the problem I have described. 
The tests even only cover good cases, which is why it is not clear. 
There are nor retrys in the tests.. 

So pls reopen that issue and give someone the change to answer the question seriously.

scaprile, 2024-04-05:
That is as far as what we support, as that is as far as what is implemented in our current API.
The unit tests show how to deal with that. As can be seen, you can implement missing parts yourself and add the missing bits and pieces on the app side, your side.
If you are a paying customer and wish to have more support for your specific needs. please contact Cesanta and request it, we'll gladly implement what you might request.
If you wish this issue to remain open, you can reopen it yourself. For the general case, we won't add anything to the API unless it is specifically requested by a paying customer, as we've already analyzed this and decided to do so. So, if this is your case, it will be served through your contract email and this issue will serve little purpose, so IMO it has served its purpose and it is closed. You may think otherwise, though, and you are welcome to do so.
Regards

RicoBiedermann, 2024-04-05:
All right, the message helps me in so far as I then send the message to the support.

scaprile, 2024-04-05:
Retry code is still present in unit_test, only the number of retries has been set to 0.
I remember testing that (a year ago?).
Let me know if there's something wrong or missing, or the tutorial is not covering something, and we'll consider that for next doc updates run. Again, for our current user base that was more than enough and going deeper wasn't worth it at that time.

cpq, 2024-04-05:
@RicoBiedermann it is quite difficult to understand your issue. Could you construct a minimal application that clearly shows an issue, please? you can take mqtt client  from examples, strip everything unnecessary, use protocol 5, and pinpoint a failing place.

RicoBiedermann, 2024-04-08:
btw. I'am unable to reopen this issue. 
Maybe because I'm not a collaborator of this repo and a collaborator has closed that issue.

scaprile, 2024-04-08:
> btw. I'am unable to reopen this issue. Maybe because I'm not a collaborator of this repo and a collaborator has closed that issue.

Sorry about that, the OP was supposed to be able to reopen. Please take my apologies.

scaprile, 2024-04-08:
Please see the mods in #2691 linked above, we are still working on it and this is by no means what we'll implement, but it might help you with your app.
IIRC we pass message flags as they are, so you should be able to check for DUP when you receive your message. Otherwise, just let me know and I'll add that to that PR.

RicoBiedermann, 2024-04-09:
At first look, it seems to me that this could solve the ID problem. 

The message structure only allows me to extract the Dup flag from the entire datagram when I receive a message.
So it would theoretically be possible. Attaching the dup flag in the  structure would be a comfort feature.
https://github.com/cesanta/mongoose/blob/b3312f376106100b7a601441856dfcf6abf72387/mongoose.h#L2372-L2382

But I dont see a field when sending a message: 
https://github.com/cesanta/mongoose/blob/b3312f376106100b7a601441856dfcf6abf72387/mongoose.h#L2355-L2370

Currently only the qos and retain field are used to set the flags in the mqq header.

RicoBiedermann, 2024-04-11:
Good idea with the dup flag! I think this wiould work.
I will test it at the begining of the next week.

cpq, 2024-04-24:
@RicoBiedermann did you manage to give it a test? please let us know.

RicoBiedermann, 2024-04-25:
> @RicoBiedermann did you manage to give it a test? please let us know.

Sorry it took so long. 

We have just completed our first test. From our side we can implement QoS1 and QoS2 with the adjustments. It would be great if this came on the main branch.

RicoBiedermann, 2024-05-07:
@cpq Is it possible to get this feature in the next mongoose release?

cpq, 2024-05-07:
@RicoBiedermann yes, sure!

scaprile, 2024-05-09:
Added in HEAD, will be included in next release next week

closed 2024-05-09 by scaprile