cesanta/mongoose #2694: MacOS unit tests fail with MbedTLS and POLL, pass with SELECT

scaprile, 2024-04-08:
```

cpq, 2024-04-17:
Any pointer to the failing test?
Works fine on my local Mac workstation

scaprile, 2024-04-17:
@cpq This is the test log. Fails consistently every night:
```
Warning: jq 1.7.1 is already installed and up-to-date.
To reinstall 1.7.1, run:
  brew reinstall jq
Warning: openssl@3 3.2.1 is already installed and up-to-date.
To reinstall 3.2.1, run:
  brew reinstall openssl@3
==> Downloading https://ghcr.io/v2/homebrew/core/mbedtls/manifests/3.6.0
==> Fetching mbedtls
==> Downloading https://ghcr.io/v2/homebrew/core/mbedtls/blobs/sha256:9133411777538739ca2650304d440cc1ae16884e3b601835cf2d2429c3063f39
==> Pouring mbedtls--3.6.0.monterey.bottle.tar.gz
🍺  /usr/local/Cellar/mbedtls/3.6.0: 198 files, 13MB
24s
Run make test ASAN_OPTIONS= MBEDTLS=`echo /usr/local/Cellar/mbedtls*/*` OPENSSL=`echo /usr/local/Cellar/openssl*/*`
[...]
235d1  3 net.c:202:mg_listen            1 4 https://localhost:12347
235d1  3 net.c:177:mg_connect           2 -1 https://localhost:12347
235d1  3 sock.c:386:mg_connect_resolved 2 5 -> 127.0.0.1:12347 pend
235d1  3 tls_mbed.c:107:mg_tls_init     2 Setting TLS
235d1  3 tls_mbed.c:135:mg_tls_init     2 hostname verification: localhost
235d4  1 event.c:31:mg_error            2 5 TLS handshake: -0x6c00
235d4  3 net.c:151:mg_close_conn        2 5 closed
235d4  3 sock.c:441:accept_conn         3 5 accepted 127.0.0.1:49190 -> 127.0.0.1:12347
235d4  3 tls_mbed.c:107:mg_tls_init     3 Setting TLS
235d5  3 sock.c:296:read_conn           3 5 0:0:0 -1 err 25
235d5  3 net.c:151:mg_close_conn        3 5 closed
FAILURE test/unit_test.c:1191: fetch(&mgr, buf, url, "GET /a.txt HTTP/1.0\n\n") == 200
```

scaprile, 2024-04-30:
While cleaning up str.h, I noticed a failure on my machine, maybe it shines some light. I added an MG_INFO to show the path taken in mg_tls_init():
```
15e1c930 3 net.c:202:mg_listen          1 4 https://localhost:12347
15e1c930 3 net.c:177:mg_connect         2 -1 https://localhost:12347
15e1c930 3 sock.c:386:mg_connect_resolv 2 5 -> 127.0.0.1:12347 pend
15e1c930 3 tls_mbed.c:107:mg_tls_init   2 Setting TLS
15e1c930 2 tls_mbed.c:131:mg_tls_init   LOAD CERT
15e1c930 3 tls_mbed.c:137:mg_tls_init   2 hostname verification: localhost
15e1c931 3 sock.c:441:accept_conn       3 6 accepted 127.0.0.1:55006 -> 127.0.0.1:12347
15e1c931 3 tls_mbed.c:107:mg_tls_init   3 Setting TLS
15e1c931 2 tls_mbed.c:128:mg_tls_init   VERIFY_NONE
15e1c949 1 event.c:31:mg_error          2 5 TLS handshake: -0x2700
15e1c949 3 net.c:151:mg_close_conn      2 5 closed
15e1c949 1 event.c:31:mg_error          3 6 TLS handshake: -0x7780
15e1c949 3 net.c:151:mg_close_conn      3 6 closed
FAILURE unit_test.c:1193: fetch(&mgr, buf, url, "GET /a.txt HTTP/1.0\n\n") == 200
```

hotraygroup, 2024-05-06:
when use mbedtls 3.x  and tls1.3,  must call psa_crypto_init before tls handshake, otherwise should occur error: TLS handshake: -0x6c00

scaprile, 2024-05-07:
@hotraygroup Thank you, can you please explain why and how this is related to this issue ? We are not using PSA.
https://mbed-tls.readthedocs.io/en/latest/getting_started/psa/
Can you also explain why tests on Ubuntu 22.04 pass ?

cpq, 2024-05-08:
Yes, apparently this psa init is required for the newer versions of mbedTLS.
The reason tests pass on ubuntu - I guess - because of the differences in mbedTLS versions used on mac vs linux.

hotraygroup, 2024-05-13:
> @hotraygroup Thank you, can you please explain why and how this is related to this issue ? We are not using PSA and there is no such error in the log above. https://mbed-tls.readthedocs.io/en/latest/getting_started/psa/ Can you also explain why tests on Ubuntu 22.04 pass ?

https://github.com/Mbed-TLS/mbedtls/issues/9072#issuecomment-2084845711  

maybe this issue only in 3.6lts

a8jan, 2024-05-16:
I can confirm only Mbed-TLS 3.6 (default version via brew on macOS) was not working with mongoose (TLS handshake: -0x6c00), previous 3.5.2 worked fine.
It was fixed with changes taken from https://github.com/cesanta/mongoose/pull/2745 
Thank you!

cpq, 2024-05-17:
Apparently the root cause for it is this:
https://github.com/Mbed-TLS/mbedtls/issues/7075 : If the connection to the target server uses TLS 1.3, then MBEDTLS_SSL_VERIFY_NONE (i.e. no CA) does not work. mbedTLS still fails with "CA required".

Closing this.

closed 2024-05-17 by cpq