cesanta/mongoose #2695: Is it possible to add the mg_mqtt_unsub function?

huenrong, 2024-04-09:
- My goal is: In the MQTT client, I need to unsubscribe topic
- My actions were: I create a function
    ```c
    void mg_mqtt_unsub(struct mg_connection *c, const struct mg_mqtt_opts *opts) {
      uint8_t qos_ = opts->qos & 3;
      size_t plen = c->is_mqtt5 ? get_props_size(opts->props, opts->num_props) : 0;
      size_t len = 2 + opts->topic.len + 2 + 1 + plen;

      mg_mqtt_send_header(c, MQTT_CMD_UNSUBSCRIBE, 2, (uint32_t) len);
      if (++c->mgr->mqtt_id == 0) ++c->mgr->mqtt_id;
      mg_send_u16(c, mg_htons(c->mgr->mqtt_id));
      if (c->is_mqtt5) mg_send_mqtt_properties(c, opts->props, opts->num_props);

      mg_send_u16(c, mg_htons((uint16_t) opts->topic.len));
      mg_send(c, opts->topic.ptr, opts->topic.len);
      mg_send(c, &qos_, sizeof(qos_));
    }
    ```
- My expectation was: unsubscribe topic
- The result I saw: Successfully unsubscribed topic
- My question is: Can this function be added to the release version?

# Environment

- mongoose version: 7.13
- Compiler/IDE and SDK: mips-linux-gnu-gcc version 7.2.0
- Target hardware/board: ingenic X2600
- Target RTOS/OS (if applicable): Linux buildroot

scaprile, 2024-04-09:
Please search, this has already been asked two weeks ago
https://github.com/cesanta/mongoose/discussions/2662#discussioncomment-8901289
Since you opened an issue instead of asking a question, is there an actual embedded issue with an embedded use case that could be satisfied with this addition ? What would that use case be ?

huenrong, 2024-04-10:
> Please search, this has already been asked two weeks ago [#2662 (comment)](https://github.com/cesanta/mongoose/discussions/2662#discussioncomment-8901289) Since you opened an issue instead of asking a question, is there an actual embedded issue with an embedded use case that could be satisfied with this addition ? What would that use case be ?

Sorry

My usage scenario is: I get some device ID that need to be subscribed through mongoose HTTP, and subscribe to the topics of these devices through MQTT. But the devices that need to be subscribed change, so I need to unsubscribe

I used mg_mqtt_unsub function in my project and it works, but I didn't write the test case, if needed, I can try it