cesanta/mongoose #2703: how can we access fn_data from mg_http_listen's callback funtion (mg_event_handler_t) ?

ffreality, 2024-04-11:
I created a wrapper for Unreal Engine 5 and exposed all features to blueprints but in order to do that, I used `fn_data` to access / cast Unreal's actor class like this. But you deleted signature and this C type callback. So, we can't use lambda capture.
How can we access fn_data ?

If there is no way, you know there are people who use mongoose with C++ projects, right ?

```
auto Callback_Listen = [](struct mg_connection* Connection, int ev, void* ev_data, void* fn_data)
{
   FHTTP_Server_Mongoose_Actor* Owner = (FHTTP_Server_Mongoose_Actor*)fn_data;
}

mg_http_listen(&this->Mongoose_Manager, TCHAR_TO_UTF8(*this->Server_Address_HTTP), Callback_Listen, (void*)this);
```

ffreality, 2024-04-11:
okay, you put it inside to struct mg_connection* Connection...

closed 2024-04-11 by ffreality

scaprile, 2024-04-11:
It has always been there
https://mongoose.ws/documentation/#struct-mg_connection
and its usage is documented
https://mongoose.ws/documentation/#mg_http_listen

ffreality, 2024-04-11:
yep, found it works as expected !