cesanta/mongoose #2724: Cleanup string API

cpq, 2024-04-24:
Please cleanup str.h to the following.
Please let me know if you have any comments.

```c
#pragma once

#include "arch.h"

// Describes an arbitrary chunk of memory
struct mg_str {
  char *buf;   // String data
  size_t len;  // String length
};

// Using macro to avoid shadowing C++ struct constructor, see #1298
#define mg_str(s) mg_str_s(s)

struct mg_str mg_str(const char *s);
struct mg_str mg_str_n(const char *s, size_t n);
int mg_casecmp(struct mg_str a, struct mg_str b);
int mg_strcmp(const struct mg_str a, const struct mg_str b);
bool mg_match(struct mg_str str, struct mg_str pattern, struct mg_str *caps);
bool mg_span(struct mg_str s, struct mg_str *a, struct mg_str *b, char delim);
uint8_t mg_toi(uint8_t c, int base);  // '7' -> 7, 'f' -> 15, etc
bool mg_path_is_sane(const struct mg_str path);
```

scaprile, 2024-04-25:
`int mg_vcasecmp(const struct mg_str *str1, const char *str2)` is used in the SSDP example. We lose the ability to do case comparison between `mg_str`s as we keep `mg_casecmp(const char *s1, const char *s2)` and `mg_strcmp(const struct mg_str str1, const struct mg_str str2)` but there is no `mg_strcasecmp`

`struct mg_str mg_strdup(const struct mg_str s)` is used by the MQTT server example AND for FreeRTOS, RTX, and Zephyr archs.  Should we write an inline hack for those ?

Some of the other functions to be removed/hidden have cross-module usage. Moving them to static will work for regular (amalgamated) builds but unamalgamated builds (tests) will fail.

cpq, 2024-04-30:
If examples use some functions, then examples must be refactored not to use them

cpq, 2024-04-30:
```c
struct mg_str a = mg_str("a");
const char *b = "b";
...
mg_vcmp(&a, b);  // before
mg_strcmp(a, mg_str(b));  // after
```

Same with globmatch

scaprile, 2024-05-01:
@cpq I think `uint8_t mg_toi(uint8_t c, int base)` serves little purpose by itself, IIUC correctly this signature. 
Our use cases are:
- HTTP:
  - turn %xx to a byte
  - decode chunk length (arbitrary hex number with 1 to 2*sizeof(int) digits) to an (unsigned?) int)
- JSON:
  - turn a JSON-encoded hex number to a byte array (apparently this is an even number of digits) (gethex)
  - unescape \uXXXX into integers
- net.c:
  - aton6()  (arbitrary hex number to an unsigned long)

We have `mg_base64_encode()` and `mg_base64_decode()`. I think it makes sense to have the base-16 equivalents, either we name then `base16`or `hex`, and the signature should be the same.
I've been using mg_snprint("%M", mg_print_hex, len, buf) so far for url_encode and others; but there is no decode equivalent. If we have an encode/decode set for base-16, then mg_print_hex would use the encode function; and the decode function would be used by cases above
These functions would go to util.h, as they would not be str functions.
WDYT

PR #2737 (2024-05-09): cleanup 'str' API
will close #2724
closed 2024-05-09 by scaprile