cesanta/mongoose #2763: Mongoose accepts requests containing multiple differing Content-Length headers.

kenballus, 2024-05-23:
To see this for yourself,

0. Compile and run the following Mongoose HTTP server:
```C
// Copyright (c) 2020 Cesanta Software Limited
// All rights reserved
// Heavily modified by Ben Kallus.

#include <stdio.h>
#include "mongoose.h"

static int const s_debug_level = MG_LL_INFO;
static char const s_listening_address[] = "http://0.0.0.0:80";

// Handle interrupts, like Ctrl-C
static int s_signo;
static void signal_handler(int signo) {
    s_signo = signo;
}

static size_t base64_encoded_len(size_t const plaintext_len) {
    // Base64-encoding takes 3n bytes to 4n bytes.
    // Then, there's padding, so we add on a correction term.
    // Note that we do not care about '\0' terminators here.
    return 4 * (plaintext_len / 3) + (plaintext_len % 3 != 0 ? 4 : 0);
}

#define INITIAL_BUF_SIZE (65535)

// *buf: A pointer to a heap-allocated buffer                      (might be `realloc`ed)
// *buf_size: The size of buf                                      (gets adjusted if *buf is `realloc`ed)
// *buf_size_remaining: The number of free bytes at the end of buf (gets adjusted if *buf is `realloc`ed)
// src: A buffer that we'll copy into buf
// src_size: The number of bytes to copy from src to buf.
void add_to_buffer(char **const buf_ptr, size_t *const buf_size_ptr, size_t *const buf_size_remaining_ptr, char const *const src, size_t const src_size) {
    char *buf = *buf_ptr;
    size_t buf_size = *buf_size_ptr;
    size_t buf_size_remaining = *buf_size_remaining_ptr;
    while (buf_size_remaining < src_size) {                              // Is there enough room?
        char *const new_buf = realloc(buf, buf_size + INITIAL_BUF_SIZE); // If not, make buf INITIAL_BUF_SIZE bigger.
        if (new_buf == NULL) {                                           // Did it work?
            puts("Out of memory!");
            exit(1);                                                     // If not, fail.
        }
        buf = new_buf;                                                   // Update buf and friends.
        buf_size += INITIAL_BUF_SIZE;
        buf_size_remaining += INITIAL_BUF_SIZE;
    }
    memcpy(buf + (buf_size - buf_size_remaining), src, src_size);        // Copy the data into the buffer
    buf_size_remaining -= src_size;

    *buf_ptr = buf;                                                      // Fill the in-out args
    *buf_size_ptr = buf_size;
    *buf_size_remaining_ptr = buf_size_remaining;
}

static char B64_SCRATCH_SPACE[INITIAL_BUF_SIZE];
static char STRCAT_SCRATCH_SPACE[INITIAL_BUF_SIZE];
static char const PAYLOAD_START[] =              "{\"headers\":[";
static char const PAYLOAD_FIRST_HEADER_START[] = "[\"";
static char const PAYLOAD_HEADER_START[] =       ",[\"";
static char const PAYLOAD_HEADER_MID[] =         "\",\"";
static char const PAYLOAD_HEADER_END[] =         "\"]";
static char const PAYLOAD_BODY_BEGIN[] =         "],\"body\":\"";
static char const PAYLOAD_URI_BEGIN[] =          "\",\"uri\":\"";
static char const PAYLOAD_METHOD_BEGIN[] =       "\",\"method\":\"";
static char const PAYLOAD_VERSION_BEGIN[] =      "\",\"version\":\"";
static char const PAYLOAD_END[] =                "\"}";
static void cb(struct mg_connection *c, int ev, void *ev_data) {
    if (ev != MG_EV_HTTP_MSG) {
[... 124 lines cut ...]
    MG_INFO(("Mongoose version : v%s", MG_VERSION));
    MG_INFO(("Listening on     : %s", s_listening_address));
    while (s_signo == 0) {
        mg_mgr_poll(&mgr, 1000);
    }
    mg_mgr_free(&mgr);
    MG_INFO(("Exiting on signal %d", s_signo));
    return 0;
}
```

1. Run the following to send it a request with two `Content-Length` headers, and observe that Mongoose does not reject the request, and prioritizes the first over the second:
```sh
printf 'POST / HTTP/1.1\r\nHost: whatever\r\nContent-Length: 34\r\nContent-Length: 0\r\n\r\nGET / HTTP/1.1\r\nHost: whatever\r\n\r\n' | nc localhost 80
```

That Mongoose prioritizes the first over the second is clear from the fact that the server responds only once. If the values of the two CL headers are swapped, then it will respond twice. This is unexpected because messages with multiple differeng Content-Length headers are disallowed by the RFCs because of request smuggling concerns.

Ideally, the fix here would be to reject messages that contain multiple Content-Length headers. This is what most HTTP servers do.

# Environment

- mongoose version: master (6bb40e6ec96f95bfd36a816b430ea2726fac9d05)
- Compiler: GCC 13.2.0
- Platform: AMD64 Linux

cpq, 2024-06-17:
We won't fix this. If the client is broken, we can't help it.

closed 2024-06-17 by cpq

kenballus, 2024-06-17:
This isn't about broken clients; it's about broken gateways. If you accept requests with conflicting CL headers, then you don't know which one the gateway prioritizes. Therefore, the only correct thing to do is reject the message.

If Mongoose continues to instead prioritize the last one arbitrarily, then Mongoose is rendered vulnerable to request smuggling when the gateway prioritized the first one, but forwarded both.

I am aware of a few gateway servers that demonstrate this behavior.

cpq, 2024-06-17:
I believe Mongoose always chooses the first content-length.
I really don't see why mongoose needs to add an extra checks in its code to fix broken gateways.

cpq, 2024-06-18:
@kenballus That makes sense, thank you - the only part I did not get is this: 

> Thus, because Mongoose doesn't reject messages with multiple conflicting Content-Length headers, Mongoose is also "broken." The developers of a gateway that forwards such invalid messages could just as easily say "we don't need to add an extra check to account for broken origin servers."

I don't see how a developers of a gateway may see messages from Mongoose with multiple Content-Length. IMO, the only place were Mongoose generates messages with Content-Length, are those for static files, and that codepath is OK. Could you give an example please?

kenballus, 2024-06-18:
You're saying that you don't need to add a check to reject messages with multiple CL headers because developers of gateways should ensure that they don't send messages containing multiple CL headers.

The wrinkle in that argument is that developers of gateways can just as easily say that *they* don't need to add an extra check to ensure that outbound messages have <=1 CL value because it's the job of origin servers to reject all messages with multiple Content-Length values.

This is circular. Both parties need to ensure that
1. such messages are both rejected, and
2. such messages are never produced.

kenballus, 2024-06-19:
> Could you clarify please - maybe a simple case where Mongoose comes broken?

I guess I'm not certain what you're asking here. As far as I'm concerned, "broken" is defined by the standards. Maybe we have different working definitions of "broken?"