cesanta/mongoose #2782: Issue with `mg_snprintf` Not Supporting `%zu` Format Specifier

TwinIsland, 2024-06-11:
- My goal is: use Add the %zu format specifier in the format string of mg_snprintf.
- My actions were: Pass a size_t variable to be formatted.
- My expectation was: should work
- The result I saw: not working, just directly render %zu instead of substituting with size_t target

# Environment

- mongoose version: 7.14
- Compiler/IDE and SDK: gcc version 13.1.0
- Target hardware/board: x86 

## Code
```C
        index += mg_snprintf(json+index, buffer_size-index, "{%m: %d, %m: %zu",
                          MG_ESC("year"), archive->year, 
                          MG_ESC("articleCount"), 12);
```

output:
```
[{"year": 2024, "articleCount": %zu"
```

TwinIsland, 2024-06-11:
use `%d` solve it

closed 2024-06-11 by TwinIsland

scaprile, 2024-06-11:
This function works as documented
https://mongoose.ws/documentation/#mg_snprintf-mg_vsnprintf

Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).