cesanta/mongoose #2785: Mongoose using OpenSSL memory leak

pmeskoni, 2024-06-14:
When mg_tls_init() provides bad data in the mg_tls_opts structure, OpenSSL contexts are allocated and failures can occur. The logic in mg_tls_init() does not clean up memory.

# Issue
On an OpenSSL failure, the following happens:
 - mg_tls_init() is called with bad data
 - some TLS failure is detected
 - mg_error() is called at all failure points in mg_tls_init(), this sets is_closing on the connection
 - goto fail within mg_tls_init()
 - free(tls) - this frees the memory for the allocated mg_tls structure, but does not free anything allocated by OpenSSL (LEAK)
 - The connection is later closed, calling mg_close_conn()
 - c->tls is still null because we never executed the code in mg_tls_init() that set it, and it has been freed anyway, so no OpenSSL data is cleaned

# Recommendation

In mg_tls_init() for OpenSSL (this would be consistent with mbedTLS implementation):
 - fail:
 -   c->tls = tls;
 -   mg_tls_free(c);

# Environment
- mongoose versions: 7.8 - 7.14, built using OpenSSL
- Compiler/IDE and SDK: OpenSSL
- Target hardware/board: any

scaprile, 2024-06-14:
### Can you please clearly explain what you mean by 
> When mg_tls_init() provides bad data

- What is "bad data" ? 
- How can we possibly "provide bad data " ?

> On an OpenSSL failure, the following happens:

- What is "an OpenSSL failure" ? Failure where ? Failure to do what ? Failure due to what ?

Then, why did you not notice that in our log ? Isn0t a clear indication there ?

### Can you please provide links to the source code where what you say is happening, apparently happens ?

Thank you.

pmeskoni, 2024-06-14:
I am currently using 7.11, but I see no changes in this area for 7.14.

I am intentionally passing junk for our testing:
 -  ca = "text:///cacert"
 - cert = ""
 - certKey = ""

This results in SSL_set_verify() failing because ca is not a proper file URI. This occurs after:
 - SSL_CTX_new() has allocated memory
 - SSL_new() has allocated memory

Do you disagree that there are paths in mg_tls_init() that can result in this memory leak?
If the OpenSSL calls cannot fail, why check for errors at all in mg_tls_init()?
Any place in mg_tls_init() that executes "goto fail;" could potentially leak memory.

scaprile, 2024-06-14:
> I am intentionally passing junk for our testing:

We don't claim compliance to any safety, availability or RFCs.
We don't cover every possible way in which the user may do things wrong, it is not our aim.

If you belong to a research group or are a paying customer and need something like that, please contact Support as per contract.
I'll gladly re-open this issue, then.
Thank you.

closed 2024-06-14 by scaprile

pmeskoni, 2024-06-14:
Just as a follow up, the Mongoose mbedTLS implementation correctly does exactly what I suggested for the Mongoose OpenSSL implementation.