cesanta/mongoose #2821: WolfSSL fails on MacOS

scaprile, 2024-07-08:
Probably a new release, 1.3 enforced ?

```
961c1  3 net.c:199:mg_listen            1 3 https://localhost:12347
961c1  3 net.c:174:mg_connect           2 -1 https://localhost:12347
961c1  3 sock.c:391:mg_connect_resolved 2 4 -> 127.0.0.1:12347 pend
961c1  3 tls_openssl.c:137:mg_tls_init  2 Setting TLS
961c2  3 tls_openssl.c:232:mg_tls_init  2 SSL client OK
961c3  3 sock.c:446:accept_conn         3 5 accepted 127.0.0.1:49197 -> 127.0.0.1:12347
961c3  3 tls_openssl.c:137:mg_tls_init  3 Setting TLS
961c3  1 event.c:31:mg_error            3 5 SSL_new
961c5  3 net.c:148:mg_close_conn        3 5 closed
961c5  3 sock.c:301:read_conn           2 4 21:0:0 -2 err 0
961c5  3 sock.c:301:read_conn           2 4 21:0:0 -2 err 0
961c5  3 net.c:148:mg_close_conn        2 4 closed
FAILURE unit_test.c:1278: fetch(&mgr, buf, url, "GET /a.txt HTTP/1.0\n\n") == 200
```

gmholland, 2024-09-11:
I have also seen this error with WolfSSL 5.6.3 but on Linux. I have not tested on macOS. The workaround was to ensure a cert and key were set on the SSL context using `SSL_CTX_use_certificate(tls->ctx, cert)` and `SSL_CTX_use_PrivateKey(tls->ctx, key)` before calling `SSL_new()`

gvanem, 2025-06-26:
@scaprile The log is similar to the above:
```
4d276b86 3 tls_openssl.c:138:mg_tls_ini 2 Setting TLS
4d276bd4 3 tls_openssl.c:236:mg_tls_ini 2 SSL client OK
4d276bd4 3 sock.c:477:accept_conn       3 604 accepted 127.0.0.1:49221 -> 127.0.0.1:12347
4d276bd4 3 tls_openssl.c:138:mg_tls_ini 3 Setting TLS
4d276bd4 1 event.c:30:mg_error          3 604 SSL_new
4d276be4 3 net.c:147:mg_close_conn      3 604 closed
4d276be4 3 sock.c:328:read_conn         2 620 21:0:0 -1 err 10054
4d276be4 3 net.c:147:mg_close_conn      2 620 closed
```

I built everything with `-DMG_ENABLE_POLL=1`. But the same timeout w/o it.