cesanta/mongoose #2822: Possible buffer overlow in S390 with TLS_BUILTIN

scaprile, 2024-07-08:
```
docker run --platform linux/amd64 --rm -e Tmp=. -e WINEDEBUG=-all -v /home/runner/work/mongoose/mongoose/test/..:/home/runner/work/mongoose/mongoose/test/.. -w /home/runner/work/mongoose/mongoose/test mdashnet/s390 cc mongoose.c unit_test.c packed_fs.c -O3 -g3  -Wmissing-prototypes -Wstrict-prototypes -pedantic -W -Wall -Werror -Wshadow -Wdouble-promotion -fno-common -Wconversion -Wundef -Isrc -I. -DMG_MAX_HTTP_HEADERS=7 -DMG_ENABLE_LINES -DMG_ENABLE_PACKED_FS=1 -DMG_ENABLE_SSI=1 -DMG_ENABLE_ASSERT=1 -DMG_ENABLE_IPV6=0  -pthread -DMG_TLS=MG_TLS_BUILTIN  -o unit_test
src/tls_chacha20.c: In function 'poly1305_update':
src/tls_chacha20.c:1237:62: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
src/tls_chacha20.c:995:17: note: at offset 16 into destination object 'buffer' of size 16
src/tls_chacha20.c:1237:62: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
src/tls_chacha20.c:995:17: note: at offset 17 into destination object 'buffer' of size 16
src/tls_chacha20.c:1237:62: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
src/tls_chacha20.c:995:17: note: at offset 18 into destination object 'buffer' of size 16
src/tls_chacha20.c:1237:62: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
src/tls_chacha20.c:995:17: note: at offset 19 into destination object 'buffer' of size 16
src/tls_chacha20.c:1237:62: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
src/tls_chacha20.c:995:17: note: at offset 20 into destination object 'buffer' of size 16
src/tls_chacha20.c:1237:62: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
src/tls_chacha20.c:995:17: note: at offset 21 into destination object 'buffer' of size 16
src/tls_chacha20.c:1237:62: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
src/tls_chacha20.c:995:17: note: at offset 22 into destination object 'buffer' of size 16
cc1: all warnings being treated as errors
```

cpq, 2025-04-11:
Is this thing still relevant ?

scaprile, 2025-04-11:
It was last time I checked, so I would answer "yes"

scaprile, 2025-07-04:
Apparently a false positive at an old compiler check. Turned that option off, memory leaks and buffer overflows are checked on several other platforms.

closed 2025-07-04 by scaprile