cesanta/mongoose #2841: Websocket server memory usage & stability

dogada, 2024-07-23:
- My goal is:  use Mongoose WS-server  to accept websocket clients and send them ESP-NOW packets.

- My actions were: use https://mongoose.ws/documentation/tutorials/websocket/websocket-server/ as base, add ESP-NOW listener and route. each ESP-NOW packet to all connected Websocket clients.
- My expectation was: ESP-NOW packets are sent without much memory usage (size of each ESP-NOW packet is 250 bytes)
- The result I saw: Mongoose io buffer increased to 80Kb of memory and identical resize errors are printed dozens on time to log, example
```
17d2c 1 mongoose.c:3392:mg_iobuf_resiz 86016->88064
17d4a 1 mongoose.c:3392:mg_iobuf_resiz 86016->88064
17d68 1 mongoose.c:3392:mg_iobuf_resiz 86016->88064
17d86 1 mongoose.c:3392:mg_iobuf_resiz 86016->88064
17db8 1 mongoose.c:3392:mg_iobuf_resiz 86016->88064
17dd6 1 mongoose.c:3392:mg_iobuf_resiz 86016->88064
17dea 1 mongoose.c:3392:mg_iobuf_resiz 86016->88064
17df4 1 mongoose.c:3392:mg_iobuf_resiz 86016->88064
```

Packets are routed to web-socket clients but periodically connection is dropped:
```
28b784 3 mongoose.c:7555:write_conn     99 55 snd 11576/12288 rcv 0/2048 n=5760 err=0
28b7e8 3 mongoose.c:7555:write_conn     99 55 snd 7361/8192 rcv 0/2048 n=5760 err=0
28b7fc 3 mongoose.c:7555:write_conn     99 55 snd 1855/2048 rcv 0/2048 n=1855 err=0
28bbee 3 mongoose.c:7555:write_conn     99 55 snd 11650/12288 rcv 0/2048 n=5760 err=0
28bc52 3 mongoose.c:7555:write_conn     99 55 snd 7365/8192 rcv 0/2048 n=5760 err=0
28bc5c 3 mongoose.c:7555:write_conn     99 55 snd 1859/8192 rcv 0/2048 n=1859 err=0
28bc66 3 mongoose.c:7545:read_conn      99 55 254:0:0 32 err 0
28bc66 3 mongoose.c:16668:mg_ws_cb      99 WS CLOSE
28bc66 3 mongoose.c:7545:read_conn      99 55 282:0:0 -1 err 128
28bc70 3 mongoose.c:7555:write_conn     99 55 snd 282/2048 rcv 0/2048 n=282 err=0
28bc70 3 mongoose.c:4778:mg_close_conn  99 55 closed
```

- My question is: 
Looks like Mongoose can't resize the memory buffer, but I suppose such huge buffer should not be used for just resending 250 bytes packets even if they arrive at high rate like 200 packets per second. 

Any suggestions to avoid such extensive memory usage and improve stability? 

The code that routes packets are pretty simple:

```

void esp_now_recv_cb(const esp_now_recv_info_t *recv_info, const uint8_t *data, int len)
{
  for (struct mg_connection *c = mgr.conns; c != NULL; c = c->next)
  {
    if (c->data[0] != 'W')
      continue;
    mg_ws_send(c, data, len, WEBSOCKET_OP_BINARY);
  }
}
```

# Environment

- mongoose version: _7.14_
- Compiler/IDE and SDK: ESP-IDF _v5.2.2_
- Target hardware/board: ESP32C3
- Connectivity chip/module: wifi:mode : sta, net80211, phy_version 1170,f4aea9b,Apr 30 2024,10:49:24
- Target RTOS/OS (if applicable):

scaprile, 2024-07-23:
Our memory usage is fine
Our stability is fine
Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).
Please read https://mongoose.ws/documentation/#best-practices
WS -> HTTP -> TCP
TCP is buffered until the manager runs. The manager sends, not the "send". If you don't call the manager, it can not send. All Mongoose functions in the same thread.

closed 2024-07-23 by scaprile

scaprile, 2024-07-24:
Your mg_ws_send() calls will fill TCP buffers until the call to mg_mgr_poll() pushes that into the OS socket buffers. If for some reason TCP does not drain those, data will stay in TCP buffers.
The manager runs when you call it, do it often, reduce the sleep timeout if necessary.
Each time data is sent, you get an MG_EV_WRITE event, you can use it to trim buffer usage.