cesanta/mongoose #2854: POST with "Transfer-Encoding: chunked" hangs

ab2022, 2024-08-01:
- My goal is: 

To process POST requests with "Transfer-Encoding: chunked" header with the tutorial
[file-upload-single-post](https://mongoose.ws/documentation/tutorials/http/file-uploads/#binary-upload-single-post)

- My actions were: 

Run the example binary in the tutorial dir on localhost. then issue this command:

`$ curl -v -H "Transfer-Encoding: chunked" -H "Content-Type: text/xml" --data-binary @small.xml http://localhost:8000/upload/small_1.xml`

- My expectation was: 
  This response:
```
< HTTP/1.1 200 OK
< Content-Length: 7
<
442 ok
```
and the file small_1.xml uploaded to /tmp.

- My other expectation was:

If it's not supported, server should respond with code 411, `"Content length missing from request"`

- The result I saw: 

The curl command hangs forever, server doesn't respond. 

- My question is: 

How can chunk transfer with POSTs be made to work?

Looking at `http_cb()` it seems like a POST with "Transfer-Encoding: chunked" should work. There is a check for that header and if found, `is_chunked` is set to true. Then there is a block `if(is_chunked)` for processing chunks.

The problem is above that in `mg_call()` line 3238. This calls main.c and there in `handle_uploads()` line 51 the headers are deleted. So `http_cb()` is checking for headers that have been stripped off.

Note also there is a typo on the tutorial web page.

`curl -H Expect: --data-binary @filename http://localhost:8000/upload?name=myfile`

fails. should be:

`curl -H Expect: --data-binary @filename http://localhost:8000/upload/myfile`

# Environment

mongoose version. git pull commit id d763fce8
Compiler/IDE: gcc 13.2.0/vscode
Target: Ubuntu 24.04
Connectivity: linux kernel 6.8.0.39 tcp/ip
Target RTOS: N/A

scaprile, 2024-08-01:
That tutorial does not use `Transfer-Encoding: chunked`.
The code has not been written to parse `Transfer-Encoding: chunked`.
The reference to the word "chunked" in the tutorial is in the context of a note, expressing the fact that this method requires the client to pace the transfer by sending spaced out "chunks" or limiting the amount of data sent at a given time. And then links to this other example https://github.com/cesanta/mongoose/tree/master/examples/file-transfer
The code in the tutorial works perfectly as it is.
If you want to use `Transfer-Encoding: chunked`, you have to write different code. We removed that example as we don't recommend using it anymore and, whatever support is left for it, could be deprecated.

What you refer to as a typo, is a change in behavior that we missed to update in this tutorial. Thank you.

closed 2024-08-01 by scaprile

ab2022, 2024-08-01:
OK thanks for info. The term "chunks" maybe overloaded. I will take a look at that example.

Just to clarify, when you say "The code has not been written to parse `Transfer-Encoding: chunked`" are you referring to the `main.c` in the tutorial/file-upload-single-post or are you referring to `mongoose.c`?

I will write my own main, for POST (and PUT) uploads should I expect to do all the chunk processing myself or will `http_cb()` do it in `if(is_chunked)`? If it is given the right headers

scaprile, 2024-08-02:
> The code has not been written to parse Transfer-Encoding: chunked.
The _tutorial/example_ code. (Prior sentence talks about the tutorial, though the full stop can be misleading

> We removed that example as we don't recommend using it anymore and, whatever support is left for it, could be deprecated.

We still have support for `Transfer-Encoding: chunked` and unit tests for it. including upload:
https://github.com/cesanta/mongoose/blob/d763fce8ed584530ebd1d4c15f69df05a7164379/test/unit_test.c#L2507-L2552

ab2022, 2024-08-02:
Great thanks! That's very helpful.