cesanta/mongoose #2857: ARP lookup fails when connection peer is the gateway

jstanzel, 2024-08-08:
When trying to do an HTTP request to an IP that is the current gateway IP, the ARP lookup will always fail (timeout) because the connection peer matching is in an else clause after the gateway equality check in rx_arp() (line 357 in net_builtin.c). I can confirm that the easiest fix (commenting out the else, so that the peer check is always done) works fine.

# Environment

- mongoose version: 7.14
- Compiler/IDE and SDK: GCC 11.3.1
- Target hardware/board: Cortex-M7
- Connectivity chip/module: custom
- Target RTOS/OS (if applicable): -

scaprile, 2024-08-08:
Thank you, but the `else` is there for a reason.
So, to clarify, your scenario is:
- Client on Mongoose
- Server on same host as gateway.
- Who is your DHCP server ? What is your configuration ?
We do an unsolicited ARP on the gateway, but your use of the gateway is not as a gateway but as a local host. Local host ARP is not working for the gateway.

jstanzel, 2024-08-08:
Of course, that's why it wasn't necessarily a suggestion, merely a data point stating what I've already tried and tested :-) The scenario is:
- Mongoose running on an embedded device acting as HTTP client, connected either via Ethernet or WiFi
- The DHCP server is the gateway (an access point in an isolated network) and hosts an HTTP server (management interface)
- The HTTP request is supposed to get status information from the access point by connecting to the management interface

PR #2860 (2024-08-12): fix ARP resolution when remote host is the gw
fixes #2857
closed 2024-08-12 by scaprile