cesanta/mongoose #2858: Simple UDP Client not working

theom4, 2024-08-10:
I want to create a UDP client which sends packets periodically. Unfortunately the board logs "network down" and the server won't receive anything I followed the docs and wrote this :
```
static void mg_event_handler(struct mg_connection* conn, int event_id, void* pvData)
{
 mg_send(udp_conn, "hello from mg!", 14);mg_event_handler, NULL);
}
```
...
struct mg_mgr mgr = {0};
  mg_log_set(MG_LL_DEBUG);
  mg_mgr_init(&mgr);
mg_connect(&mgr,"udp://192.168.100.67:4023", mg_event_handler, NULL);
 while (1)
  {
    mg_mgr_poll(&mgr,1000);
}

- mongoose version: 7.14
- Compiler/IDE and SDK: VS Code with STM32 Extension
- Target hardware/board: Nucleo-F756ZG
- No OS
[image]

scaprile, 2024-08-10:
Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).
https://mongoose.ws/documentation/#udp
I don't see we do something like that in any of our examples. No event handler is like that. (That wouldn't even build, in fact)
Besides that, your link is not coming up. Your PHY is not seeing your Ethernet cable is connected. Is it ?
We don't have any examples for the F756, what makes you think you can just drop some code and it will just work out of the blue ? It is even not listed as supported. We do have F756 code in our Wizard: https://mongoose.ws/wizard/#/output?board=f756&ide=GCC+make&rtos=baremetal&file=README.md

closed 2024-08-10 by scaprile

theom4, 2024-08-11:
Ah, ignore the code after the ";" in the handler. I followed the documentation and the only APIs I have seen available for plain UDP/ TCP client are ``mg_send()`` and ``mg_connect()``. I'm not an expert to understand the UDP examples, I want to implement only a simple client. A UDP server is working perfectly fine on my side.

scaprile, 2024-08-11:
I hope you'll understand that this is not an issue of ours. You clearly did not follow the "events" section of the docs.
Examples are not for experts but for basic user level; I even pointed you to a generator that will craft working code for you, where you just need to add two basic lines of code taken from extremely basic examples; those that I just linked you to, too.

Besides that, I just told you that 
> your link is not coming up. Your PHY is not seeing your Ethernet cable is connected. Is it ?

Do you actually understand what that means ? Before your correct UDP code is called, the link has to be UP, and it IS NOT. The code generator I pointed you to, will generate correct code for a Nucleo F756 board. Then the link should come up.
Your code is not handling events, it will do a lot of things that it must not do. Your server code is probably also wrong, but as it is a server it is masking your mistakes. Please see the examples, understand the examples, or at least copy what they do.
The other closest thing we have to a UDP client is the SNTP client (SNTP runs over UDP), that code is automatically generated by the wizard, try to adapt that to your needs if you are unable to understand our simplest possible UDP examples that I linked you to.

If you need further support, and are a paying customer, please contact Support as per your contract.