cesanta/mongoose #2866: Mongoose 7.1x: Random UDP Port Listening Issue with mg_tls_init

mmmkkk001, 2024-08-22:
My goal is:
To use TLS-related functionality in the mongoose library without the program automatically listening on a random UDP port.

My actions were:
I used the mongoose library and called the mg_tls_init(c, &opts); function in my code. I did not enable the -DMG_TLS=MG_TLS_BUILTIN option during compilation.

My expectation was:
I expected the program to initialize TLS without automatically listening on any UDP ports, as this is not needed for my application.

The result I saw:
Every time I compile and run the program, it automatically listens on a random UDP port whenever the mg_tls_init(c, &opts); function is called. This happens even though I did not enable the -DMG_TLS=MG_TLS_BUILTIN option.

My question is:
Why does calling mg_tls_init(c, &opts); cause the program to listen on a random UDP port, and how can I prevent this from happening?

Environment:

mongoose version: mongoose-7.15
Compiler/IDE and SDK: GCC version 14.2.0 (GCC), GCC version 13.2.1 20240309
Target RTOS/OS (if applicable): Linux

scaprile, 2024-08-22:
Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).

It is impossible to guess what you are doing wrong if you don't say what you do

mmmkkk001, 2024-08-23:
I used the Mongoose library to create a WebSocket client connection with the following code:
`#include "mongoose.h"

static const char *s_url = "ws://589.jyhdkj.com";

// Print websocket response and signal that we're done
static void fn(struct mg_connection *c, int ev, void *ev_data) {
  if (ev == MG_EV_OPEN) {
    c->is_hexdumping = 1;
  } else if (ev == MG_EV_ERROR) {
    // On error, log error message
    MG_ERROR(("%p %s", c->fd, (char *) ev_data));
  } else if (ev == MG_EV_WS_OPEN) {
    // When websocket handshake is successful, send message
    mg_ws_send(c, "hello", 5, WEBSOCKET_OP_TEXT);
  } else if (ev == MG_EV_WS_MSG) {
    // When we get echo response, print it
    struct mg_ws_message *wm = (struct mg_ws_message *) ev_data;
    printf("GOT ECHO REPLY: [%.*s]\n", (int) wm->data.len, wm->data.buf);
  }

  if (ev == MG_EV_ERROR || ev == MG_EV_CLOSE) {
    *(bool *) c->fn_data = true;  // Signal that we're done
  }
}

int main(void) {
  struct mg_mgr mgr;        // Event manager
  bool done = false;        // Event handler flips it to true
  struct mg_connection *c;  // Client connection
  mg_mgr_init(&mgr);        // Initialise event manager
  mg_log_set(MG_LL_DEBUG);  // Set log level
  c = mg_ws_connect(&mgr, s_url, fn, &done, NULL);     // Create client
  while (c && done == false) mg_mgr_poll(&mgr, 1000);  // Wait for echo
  mg_mgr_free(&mgr);                                   // Deallocate resources
  return 0;
}`
What I observed:
After connecting to the WebSocket server, I noticed that the UDP port temporarily opened by the sendto function during the DNS query remains open for the entire duration of the WebSocket connection. The port does not automatically close, even after the DNS resolution is complete.

My question is:
Why does the UDP port opened by the sendto function remain open for the entire connection duration? Is there a way to ensure that this port is closed once the DNS resolution is complete, especially when the WebSocket connection is being maintained?

mmmkkk001, 2024-08-23:
[image]
Image 1 shows a successful connection to the WebSocket server, with the connection remaining open.  

[image]
Image 2 shows the UDP port that was opened simultaneously. This port does not automatically close but remains open.

scaprile, 2024-08-23:
Please don't go closing and reopening issues.
Please don't open issues to ask questions
Please properly format code, it is impossible to see that
Please don't post snapshots, it is impossible to see traffic in a snapshot, attach capture files instead
There is no random UDP port and there is no relation to mg_tls_init()
Mongoose opens a UDP port to resolve DNS and does not close it. That's the way it works now. If you need/want it work differently, and you are a paying customer, please contact Cesanta as per your contract.

closed 2024-08-23 by scaprile

mmmkkk001, 2024-09-07:
The issue is that this operation is unnecessary.  After resolving the domain name, the port should be closed promptly, as it won't be needed afterward.  This is clearly a mistake.  Since the IP address of the domain has already been obtained, the UDP port should be closed immediately.Otherwise, you will see that the UDP port will remain open throughout the entire lifecycle of the program, just like in the image.
[image]

scaprile, 2024-09-07:
No one is preventing you from forking and doing what you think you can do better.
Thank you