cesanta/mongoose #2885: Mongoose as https client fails to download larger files ~ 6kB

jcorporation, 2024-09-05:
- My goal is: Use mongoose as http client to download small images
- My actions were: I implemented the http client as in the tutorial https://mongoose.ws/documentation/tutorials/http/http-client/
- My expectation was: Mongoose receives the always the complete image
- The result I saw: Mongoose reads not the complete image. I added logging for hm->body.len and the Content Length header of the HTTP response:

```
20:36:44 DEBUG    webserver       src/web_server/proxy.c:221: Got 5468 bytes from connection "72", response code 200
20:36:44 DEBUG    webserver       src/web_server/proxy.c:226: Connection "72": Content-Length: 6022
```

URL is: https://jcorporation.github.io/webradiodb/db/pics/http___as-hls-ww-live_akamaized_net_pool_904_live_ww_bbc_radio_two_bbc_radio_two_isml_bbc_radio_two-audio_3d96000_norewind_m3u8.webp

Downloading this image works with Mongoose 7.14. The failure occurs with Mongoose 7.15.

To validate that is not a bug in my code I used your tutorials/http/http-client do download the image and got the same error.
You can find attached the log and the diff of the http-client.

[log.txt](https://github.com/user-attachments/files/16899133/log.txt)
[diff.txt](https://github.com/user-attachments/files/16899134/diff.txt)

- My question is: Seems to be a bug? Do you need further logs?

# Environment

- mongoose version: 7.15
- Compiler/IDE and SDK: gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
- Target hardware/board: Linux t14 6.5.0-15-generic #15-Ubuntu SMP PREEMPT_DYNAMIC Tue Jan  9 17:03:36 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
- Connectivity chip/module: -
- Target RTOS/OS (if applicable): -

scaprile, 2024-09-06:
Hi, Jürgen, reproduced and investigating

scaprile, 2024-09-09:
@jcorporation Please test #2890 and let us know; works for me with that.

PR #2890 (2024-10-01): Fix multiple TLS records in buffer
Even though we receive IO_ERR, several outstanding TLS records can lie in our buffer. One or two extra calls is not enough to correctly process them and cleanly close.

NOTE: is_io_err becomes unused, will be removed further on if this proves to be the final battle.

fixes #2885 
closes #2668 

PS: In rare cases, rtls was filled with more than one record, and our capping strategy prevented us to call mg_tls_recv(). Fixed.
closed 2024-10-01 by scaprile