cesanta/mongoose #2887: Use OPENSSL How to make this connection timeout? to close the mg_connection

gogoxx2024, 2024-09-07:
- My goal is: Use OPENSSL
- My actions were: Connect using TLS, process received data
- My expectation was: close the test tool[restclient.jar] 
- The result I saw: There is still one link that has not been released;  static void read_conn(struct mg_connection *c) 
-
- My question is: How to make this connection timeout? to close the mg_connection

# Environment

- mongoose version: 7.15
- Compiler/IDE and SDK: vs2015
- Target hardware/board: windows11
- Connectivity chip/module: mongoose-master\tutorials\http\http-restful-server
- Target RTOS/OS (if applicable): None

line:7523
static void read_conn(struct mg_connection *c)

n is MG_IO_RESET

c->is_tls_hs is 0

if (c->is_tls_hs) mg_tls_handshake(c);

How to make this connection timeout? to close the mg_connection

[image]
[image]
[image]
close the tool
[image]
[image]
[image]

scaprile, 2024-09-07:
I don't understand what you are doing.
Please don't post snapshots, refer to code and attach network captures (files)
Please don't mark our code, we already know it, it is yours that we don't know
Please correctly describe what you are doing, what is your code, what happens, where, and why
Minimum possible to reproduce your issue.
Please read https://mongoose.ws/documentation/#best-practices
Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).
When you have that, then we'll reopen this issue if necessary

closed 2024-09-07 by scaprile

gogoxx2024, 2024-09-08:
What I want to say is that I use SSL HTTP server. When the client connects and communicates, the client closes and the server still has a connection pointer that has not been reclaimed, so it will not be released. What I want to ask is if there is any API that can handle timeouts to release such invalid links

scaprile, 2024-09-08:
> Please correctly describe what you are doing, what is your code, what happens, where, and why
> Minimum possible to reproduce your issue.
> Please read https://mongoose.ws/documentation/#best-practices [...]

gogoxx2024, 2024-09-08:
I uploaded my test code to Baidu Cloud
Link: https://pan.baidu.com/s/1l8D6Cb1of4PK24E6qpreww  
Extraction code: ol6u

HttpDemo. zip is the source code for the vs2015 code project, mongoose master \ tutorials \ http \ http critical server
openssl-x64.zip is the ssllib
Restclient. zip is a link testing tool for HTTPS
Recording. zip demo video

scaprile, 2024-09-08:
If you would have read the tutorials as instructed: https://mongoose.ws/documentation/tutorials/tls/#windows
>Windows
Only what the example provides in its Makefile is supported 

We do not support custom openssl in Windows.
Whatever you want to submit to reproduce your issue must be posted here, clear and readable code, and build as standard. As we won't build nor link any third-party library, I will dismiss this.