cesanta/mongoose #2898: `driver init failed` when using different SPI pins with RP2040-based board

TheGuyDanish, 2024-09-19:
- My goal is: To use Mongoose as a web server to present readouts/perform basic GPIO control.
- My actions were: Changed the `pico-w5500` example to use the pins assigned on my board:
- `enum { LED = 6, SPI_CS = 5, SPI_CLK = 2, SPI_TX = 3, SPI_RX = 4 };`
- My expectation was: That the board would come up and present the web server
- The result I saw:
```
2711   2 main.c:68:main                 Starting ...
2714   1 mongoose.c:5927:mg_tcpip_init  driver init failed
2719   2 main.c:90:main                 MAC: 02:74:db:3d:1b:35. Waiting for IP...
2b08   2 main.c:42:timer_fn             Ethernet: down, IP: 0.0.0.0, rx:0, tx:0, dr:0, er:0
```
- My question is: Referring to the [Pico Pinout](https://www.raspberrypi.com/documentation/microcontrollers/images/pico-pinout.svg), GP2-5 are on spi0, so I believe this should function but I might just be missing something.
The board initializes successfully using MicroPython with the following code:
```py
spi = SPI(0, 2000000, mosi=Pin(3), miso=Pin(4), sck=Pin(2))
nic = network.WIZNET5K(spi, Pin(5), Pin(0)) # Bus, CSn, RSTn
nic.active(True)
while not nic.isconnected():
  sleep(1)
```

# Environment

- mongoose version: master/`9499669`
- Compiler/IDE and SDK: GCC 11.4.0 on Ubuntu 22.04, GNU Make 4.3
- Target hardware/board: RP2040 ([Custom board schematic](https://github.com/OpenJBOD/rp2040/releases/download/rev4.1/OpenJBOD-RP2040-SCH.pdf))
- Connectivity chip/module: WIZnet W5500
- Target RTOS/OS (if applicable): N/A

scaprile, 2024-09-19:
Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).

The pinout in the example is for the board in the example.
Other possible pinouts depend on your board.
https://mongoose.ws/documentation/tutorials/rp2040/pico-w5500/
All the examples in that tutorial work as expected
Your expectation is not realistic, if you choose your board and choose your pinout, you can't expect us to guess why it does not work for you.
I don't understand what is that code you place there
I'm sorry, but I won't check your schematic. If you are a paying customer or want that service, please contact Support

closed 2024-09-19 by scaprile

TheGuyDanish, 2024-09-19:
Understood, I'll find another library. Thank you for your assistance.