cesanta/mongoose #2920: Mongoose didn't work when Data cache enable on STM32F7

LinogeA, 2024-10-02:
Hello everyone!
My own little demo project for the NUCLEO-F746ZG board stops working if I enable Data cache (D-cache) in the microcontroller. I give all other details below.

In the demo project, the microcontroller is connected at a fixed address and periodicaly transmits data in the form of UDP packets. When D-cache is enabled, the message "No free descriptors" is displayed in the terminal at each attempt to send data. I am not used here any RTOS and code is simple as possible. IDE is KEIL and was tested with different verisons of compiler.

The first option to solve the problem is to do as STMicroelectronics does: place descriptors and buffers in memory at fixed addresses and then use MPU to disable caching of these regions.

The second option to solve the problem was inspired by this thread:
https://community.st.com/t5/stm32-mcus-embedded-software/how-to-make-ethernet-and-lwip-working-on-stm32/td-p/261456

And special attention should be paid to these topics:
https://community.st.com/t5/stm32-mcus-products/maintaining-cpu-data-cache-coherence-for-dma-buffers/m-p/95746
https://community.st.com/t5/stm32-mcus-embedded-software/bug-missing-compiler-and-cpu-memory-barriers/m-p/298508
https://community.st.com/t5/stm32cubemx-mcus/lwip-v6-5-0-stmh32h735-disco-https-github-com-stm32-hotspot/m-p/91721

Next I will show where I made edits according to the above recommendations.

1. First one I force to place descriptors into DTCM RAM which is non-cacheable. Motivation was to mitigate using cache invalidation for such small elements.
`__attribute__((section(".ARM.__at_0x20000000"))) static uint32_t s_rxdesc[ETH_DESC_CNT][ETH_DS]; // RX descriptors
 __attribute__((section(".ARM.__at_0x20000064"))) static uint32_t s_txdesc[ETH_DESC_CNT][ETH_DS]; // TX descriptors
 `
2. I have changed the max frame size to align it with __SCB_DCACHE_LINE_SIZE value (1540 -> 1536).
`#define ETH_PKT_SIZE 1536`
3. Clean cache macro was added to transmit function mg_tcpip_driver_stm32f_tx:
```
static size_t mg_tcpip_driver_stm32f_tx(const void *buf, size_t len,
                                        struct mg_tcpip_if *ifp) {
if (len > sizeof(s_txbuf[s_txno])) {
    MG_ERROR(("Frame too big, %ld", (long) len));
    len = 0;  // Frame is too big
  } else if ((s_txdesc[s_txno][0] & MG_BIT(31))) {
    ifp->nerr++;
    MG_ERROR(("No free descriptors"));
    // printf("D0 %lx SR %lx\n", (long) s_txdesc[0][0], (long) ETH->DMASR);
    len = 0;  // All descriptors are busy, fail
  } else {
    memcpy(s_txbuf[s_txno], buf, len);                           // Copy data

   SCB_CleanDCache_by_Addr(s_txbuf[s_txno], len);

    s_txdesc[s_txno][1] = (uint32_t) len;                        // Set data len
    s_txdesc[s_txno][0] = MG_BIT(20) | MG_BIT(28) | MG_BIT(29);  // Chain,FS,LS
    s_txdesc[s_txno][0] |= MG_BIT(31);  // Set OWN bit - let DMA take over
    if (++s_txno >= ETH_DESC_CNT) s_txno = 0;
  }
  MG_DSB();                            // ensure descriptors have been written
  ETH->DMASR = MG_BIT(2) | MG_BIT(5);  // Clear any prior TBUS/TUS
  ETH->DMATPDR = 0;                    // and resume
  return len;
}
```
4. I added a memory barrier after checking the OWN bit and before reading the descriptor value, and put a memory barrier before setting the OWN bit.:
```
#ifdef __riscv
__attribute__((interrupt()))  // For RISCV CH32V307, which share the same MAC
#endif
void ETH_IRQHandler(void);
void ETH_IRQHandler(void) {
  if (ETH->DMASR & MG_BIT(6)) {           // Frame received, loop
    ETH->DMASR = MG_BIT(16) | MG_BIT(6);  // Clear flag
    for (uint32_t i = 0; i < 10; i++) {   // read as they arrive but not forever
      if (s_rxdesc[s_rxno][0] & MG_BIT(31)) break;  // exit when done      
	  __DMB();
	  if (((s_rxdesc[s_rxno][0] & (MG_BIT(8) | MG_BIT(9))) ==
           (MG_BIT(8) | MG_BIT(9))) &&
          !(s_rxdesc[s_rxno][0] & MG_BIT(15))) {  // skip partial/errored frames
        uint32_t len = ((s_rxdesc[s_rxno][0] >> 16) & (MG_BIT(14) - 1));
        //  printf("%lx %lu %lx %.8lx\n", s_rxno, len, s_rxdesc[s_rxno][0],
        //  ETH->DMASR);
        mg_tcpip_qwrite(s_rxbuf[s_rxno], len > 4 ? len - 4 : len, s_ifp);
[... 85 lines cut ...]
      if (++s_rxno >= ETH_DESC_CNT) s_rxno = 0;
    }
  }
  // Cleanup flags
  ETH->DMASR = MG_BIT(16)    // NIS, normal interrupt summary
               | MG_BIT(7);  // Clear possible RBUS while processing
  ETH->DMARPDR = 0;          // and resume RX
}
```
5. Final step was to add cache invalidation in function that copy data from buffer that was filled by DMA:
```
void mg_tcpip_qwrite(void *buf, size_t len, struct mg_tcpip_if *ifp) {
  char *p;
  if (mg_queue_book(&ifp->recv_queue, &p, len) >= len) {
    SCB_InvalidateDCache_by_Addr(buf, len);
    memcpy(p, buf, len);
    mg_queue_add(&ifp->recv_queue, len);
    ifp->nrecv++;
  } else {
    ifp->ndrop++;
  }
}
```

Thats all! These is how it was used in my test project. Of course, my modifications may be done in wrong places, because I am novice in subject of cache coherency. But I hope that make Mongoose better. Sorry for such long post.

scaprile, 2024-10-02:
TL:DR
Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).
We do NOT enable the cache in our examples

closed 2024-10-02 by scaprile

LinogeA, 2024-10-02:
Then this is a big oversight on your part and you should write in the documentation in big red letters that working with cache is not supported. There is no mention of working with cache on your site, you can see for yourself.