cesanta/mongoose #2937: Mongoose closes http connection before response is generated in a thread

jcorporation, 2024-10-10:
- My goal is: Generate a response in a thread and send it back to the client
- My actions were: I use an async message queue to generate the response
- My expectation was: Mongoose does not close the connection before the response is generated, even if the Connection:close header is set
- The result I saw: Mongoose closes the connection directly after calling `mg_call(c, MG_EV_HTTP_MSG, &hm);`
- My question is:

This behavior was introduced with commit 9974be3fdc149e0477dc61b95dc2ba556fe43054.

The connection is set to draining directly after calling the MG_EV_HTTP_MSG event handler if "connection: close" is found in the request header. If this event handler does not respond directly to the request, but pushes a work request to an another thread, the connection was closed by mongoose before the other thread has a chance to generate the response.

```
# New connection with id 4
web_server.c:1003: 30625bb2f 3 mongoose.c:7719:accept_conn 4 15 accepted 127.0.0.1:54920 -> 0.0.0.0:8080
web_server.c:610: New connection id "4" from 127.0.0.1, connections: 3
web_server.c:1003: 30625bb2f 4 mongoose.c:7974:mg_mgr_poll 4 r- tchrc 0 0
web_server.c:1003: 30625bb2f 4 mongoose.c:7518:recv_raw    4 506 0
web_server.c:1003: 30625bb2f 3 mongoose.c:7573:read_conn   4 15 0:0:0 506 err 0
web_server.c:668: HTTP request (4): POST /api/default
# Push the work request to another thread
request_handler.c:35: "default": API request (4): {"jsonrpc":"2.0","id":428675007,"method":"MYMPD_API_PLAYER_STATE","params":{}}
request_handler.c:56: "default": API request (4): MYMPD_API_PLAYER_STATE
mympd_api.c:181: Queue event
# Request handler returns and mongoose closes the connection
web_server.c:894: HTTP connection "4" closed
web_server.c:1003: 30625bb2f 3 mongoose.c:4785:mg_close_co 4 15 closed
# Response is generated and pushed back to the thread running the mg_mgr
mympd_api.c:192: "default": Queue event
web_server.c:1003: 30625bb32 4 mongoose.c:7974:mg_mgr_poll 2 r- tchrc 0 0
web_server.c:1003: 30625bb33 4 mongoose.c:7518:recv_raw    2 4 0
web_server.c:1003: 30625bb33 3 mongoose.c:7573:read_conn   2 11 0:0:0 4 err 0
# The connection was already removed and we can not send the response
web_server.c:259: "default": Got API response for id "4"
web_server.c:508: Connection for id "4" not found
```

# Environment

- mongoose version: 7.15
- Compiler/IDE and SDK: linux / gcc 14

scaprile, 2024-10-10:
Should've been fixed, please confirm

closed 2024-10-10 by scaprile

jcorporation, 2024-10-11:
I had to set c->is_resp in the MG_EV_HTTP_MSG and MG_EV_WS_OPEN handlers to get it working. After I looked through your patch, I think this is how it should be done.

scaprile, 2024-10-11:
c->is_resp is set before calling the event handler with the EV_HTTP_MSG event. If you reply using mg_http_reply() or mg_http_*_chunk(), that will reset c->is_resp and Mongoose will assume you finished writing your response, so, in order to keep generating a response, you either don't reply, use other functions, or set c->is_resp.

jcorporation, 2024-10-11:
Thanks, It works like I assumed.

PR #2938 (2024-10-10): Honor c->is_resp with Connection:close
fixes #2937