cesanta/mongoose #2969: HTTP client does not always read the whole response

jcorporation, 2024-11-17:
- My goal is: Use mongoose as http client with "Connection: close" header set
- My actions were: Used the http client tutorial and added the "Connection: close" header
- My expectation was: Mongoose reads the whole response
- The result I saw: Mongoose does not always read the whole response if "Connection: close" is set in the request

```
make CFLAGS_EXTRA="-DMG_TLS=MG_TLS_OPENSSL -lssl -lcrypto" ARGS="https://mongoose.ws/documentation/#mg_http_get_header"
```

Diff of the tutorial for the test:

```
diff --git a/tutorials/http/http-client/main.c b/tutorials/http/http-client/main.c
index 8aee400b..465e2c1d 100644
--- a/tutorials/http/http-client/main.c
+++ b/tutorials/http/http-client/main.c
@@ -29,8 +29,7 @@ static void fn(struct mg_connection *c, int ev, void *ev_data) {
     struct mg_str host = mg_url_host(s_url);
 
     if (mg_url_is_ssl(s_url)) {
-      struct mg_tls_opts opts = {.ca = mg_unpacked("/certs/ca.pem"),
-                                 .name = mg_url_host(s_url)};
+      struct mg_tls_opts opts = {};
       mg_tls_init(c, &opts);
     }
 
@@ -41,6 +40,7 @@ static void fn(struct mg_connection *c, int ev, void *ev_data) {
               "Host: %.*s\r\n"
               "Content-Type: octet-stream\r\n"
               "Content-Length: %d\r\n"
+              "Connection: close\r\n"
               "\r\n",
               s_post_data ? "POST" : "GET", mg_url_uri(s_url), (int) host.len,
               host.buf, content_length);
```

# Environment

- mongoose version: current master
- Compiler/IDE and SDK: gcc version 14.2.0 (Ubuntu 14.2.0-4ubuntu2)
- Target hardware/board: Linux t14 6.11.0-9-generic #9-Ubuntu SMP PREEMPT_DYNAMIC Mon Oct 14 13:19:59 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

scaprile, 2024-11-18:
I see... I can imagine, this one is tough.

jcorporation, 2024-11-18:
Yes, I tried it with different servers. It is the same behavior.

aeschimannr, 2024-11-18:
Hello, I’m encountering the same issue. When requesting RTCM data from an NTRIP caster, I receive the data in the MG_EV_READ event. However, the buffer fails to parse in the function `mongoose.c:3411: int n = mg_http_parse(buf, c->recv.len - ofs, &hm);` which returns -1.

As a result, the `MG_EV_READ` event handler receives the buffer in `c->recv.buf`, but `c->recv.len` is set to 0 and impossible to read.

I’ve observed this issue on an STM32F4 target and have included an image for reference.

 [image]

jcorporation, 2024-11-19:
Thanks for investigation.

I will test this, but in my production code I catch this already and throw an error if the content-length header exists and the received body length is smaller.

I enhance your tutorial code to illustrate the issue in more detail.

Removing the Connection: close header from the request solves the isssue immedately. This is strange but my observation.

scaprile, 2024-11-20:
There is no content-length to check against, in this case.
Mongoose can't possibly know whether there is outstanding data in flight or not. We can check valid behavior later, I need to know if this your experienced failure mechanism or we need to search for something else.

jcorporation, 2024-11-20:
For my test with https://mongoose.ws/documentation/ I checked if the output of the html is complete or not.

scaprile, 2024-11-20:
> For my test with https://mongoose.ws/documentation/ I checked if the output of the html is complete or not.

OK, but did you take the prior error event into account ?
One thing (scenario 1) is an incomplete HTTP content (based on prior knowledge of it, or syntax check, out-of-band checks anyway) delivered after an error event, which is fine (we can track that error later); but another thing (scenario 2) is the same incomplete content with no indication whatsoever. Where are we ? So far I've only seen scenario 1.

jcorporation, 2024-11-20:
Here are the two logs:

[log-fail.txt](https://github.com/user-attachments/files/17837804/log-fail.txt)
[log-ok.txt](https://github.com/user-attachments/files/17837805/log-ok.txt)

These are produced with your http-client tutorial, patched with:

```
diff --git a/tutorials/http/http-client/main.c b/tutorials/http/http-client/main.c
index 8aee400b..24510ec5 100644
--- a/tutorials/http/http-client/main.c
+++ b/tutorials/http/http-client/main.c
@@ -29,8 +29,7 @@ static void fn(struct mg_connection *c, int ev, void *ev_data) {
     struct mg_str host = mg_url_host(s_url);
 
     if (mg_url_is_ssl(s_url)) {
-      struct mg_tls_opts opts = {.ca = mg_unpacked("/certs/ca.pem"),
-                                 .name = mg_url_host(s_url)};
+      struct mg_tls_opts opts = {};
       mg_tls_init(c, &opts);
     }
 
@@ -41,6 +40,7 @@ static void fn(struct mg_connection *c, int ev, void *ev_data) {
               "Host: %.*s\r\n"
               "Content-Type: octet-stream\r\n"
               "Content-Length: %d\r\n"
+              "Connection: close\r\n"
               "\r\n",
               s_post_data ? "POST" : "GET", mg_url_uri(s_url), (int) host.len,
               host.buf, content_length);
@@ -48,10 +48,11 @@ static void fn(struct mg_connection *c, int ev, void *ev_data) {
   } else if (ev == MG_EV_HTTP_MSG) {
     // Response is received. Print it
     struct mg_http_message *hm = (struct mg_http_message *) ev_data;
-    printf("%.*s", (int) hm->message.len, hm->message.buf);
+    fprintf(stderr, "%.*s", (int) hm->message.len, hm->message.buf);
     c->is_draining = 1;        // Tell mongoose to close this connection
     *(bool *) c->fn_data = true;  // Tell event loop to stop
   } else if (ev == MG_EV_ERROR) {
+    MG_ERROR(("ERROR"));
     *(bool *) c->fn_data = true;  // Error, tell event loop to stop
   }
 }
```

scaprile, 2024-11-21:
OK, that is scenario 2.
I'm able to reproduce it, though in my case it does not depend on `Connection: close`. Let's see what I can find

scaprile, 2024-11-21:
Please try the fix in #2971, works for me.

PR #2971 (2024-11-22): Fix issuing 411 responses as a client
15.5.12. 411 Length Required

The 411 (Length Required) status code indicates that **the server** refuses to accept the request without a defined Content-Length (Section 8.6). The client MAY repeat the request if it adds a valid Content-Length header field containing the length of the request content.

fixes #2969
commit 5c305e7f (2024-11-22): Fix issuing 411 responses as a client

15.5.12. 411 Length Required

The 411 (Length Required) status code indicates that the server refuses to accept the request without a defined Content-Length (Section 8.6). The client MAY repeat the request if it adds a valid Content-Length header field containing the length of the request content.

fixes #2969
closed 2024-11-22 by scaprile