cesanta/mongoose #2980: Add the use of .hex files for firmware upgrades.

chriskinal, 2024-12-05:
- My goal is: to use the more common .hex files to perform firmware upgrades. Arduino IDE and other Arduino tools generate .hex files for firmware upgrades. 
- My actions were: attempt to use .hex file for firmware upgrade.
- My expectation was: successful firmware upgrade.
- The result I saw: .hex files can not be selected in the firmware upgrade GUI.
- My question is: can the feature be added to allow firmware upgrades using .hex files?

# Environment

- mongoose version: 7.16
- Compiler/IDE and SDK: VSCode/PlatformIO and Teensyduino SDK.
- Target hardware/board: Teensy 4.1
- Connectivity chip/module: On board.
- Target RTOS/OS (if applicable): Bare metal.

scaprile, 2024-12-05:
Sorry, this is not an issue.
If you have a feature request, then please use the proper channel: https://github.com/cesanta/mongoose/discussions/categories/feature-requests
Thank you

closed 2024-12-05 by scaprile

chriskinal, 2024-12-05:
Apologies, most repos don’t use discussions so I neglected to look there. I will raise the feature request there. Thanks!