cesanta/mongoose #3010: build web-ui-dashboard

embetrix, 2025-01-15:
following the description in : https://mongoose.ws/documentation/tutorials/device-dashboard/#build-and-run---demo

```
$ make -C reference-projects/windows-macos-linux/web-ui-dashboard/ clean all
make: Entering directory '/home/zaki/Projects/Playground/mongoose/reference-projects/windows-macos-linux/web-ui-dashboard'
rm -rf          ./example        *.o *.obj *.exe *.dSYM mbedtls
cc main.c mongoose.c net.c packed_fs.c -W -Wall -Wextra -Wundef -Wshadow -Wdouble-promotion -Wconversion -Wno-sign-conversion -fno-common -ffunction-sections -fdata-sections -g3 -Os -I. -DMG_ENABLE_PACKED_FS=1  -o ./example              
/usr/bin/ld: /tmp/ccoyJHvO.o: in function `packed_open':
/home/zaki/Projects/Playground/mongoose/reference-projects/windows-macos-linux/web-ui-dashboard/mongoose.c:1037:(.text.packed_open+0x28): undefined reference to `mg_unpack'
/usr/bin/ld: /tmp/ccoyJHvO.o: in function `packed_stat':
/home/zaki/Projects/Playground/mongoose/reference-projects/windows-macos-linux/web-ui-dashboard/mongoose.c:1006:(.text.packed_stat+0x2f): undefined reference to `mg_unpack'
/usr/bin/ld: /home/zaki/Projects/Playground/mongoose/reference-projects/windows-macos-linux/web-ui-dashboard/mongoose.c:1008:(.text.packed_stat+0x44): undefined reference to `mg_unlist'
/usr/bin/ld: /tmp/ccoyJHvO.o: in function `mg_unpacked':
/home/zaki/Projects/Playground/mongoose/reference-projects/windows-macos-linux/web-ui-dashboard/mongoose.c:993:(.text.mg_unpacked+0x24): undefined reference to `mg_unpack'
/usr/bin/ld: /tmp/ccoyJHvO.o: in function `packed_list':
/home/zaki/Projects/Playground/mongoose/reference-projects/windows-macos-linux/web-ui-dashboard/mongoose.c:1020:(.text.packed_list+0x67): undefined reference to `mg_unlist'
collect2: error: ld returned 1 exit status
make: *** [Makefile:29: example] Error 1
make: Leaving directory '/home/zaki/Projects/Playground/mongoose/reference-projects/windows-macos-linux/web-ui-dashboard

```

scaprile, 2025-01-15:
The issue template is there for a reason: to save yours and our time. Please honor the issue template and fill it as requested.
```
$ make -C reference-projects/windows-macos-linux/web-ui-dashboard/ clean all
make: Entering directory '/home/scaprile/work/cesanta/mongoose/reference-projects/windows-macos-linux/web-ui-dashboard'
rm -rf          ./example        *.o *.obj *.exe *.dSYM mbedtls
node pack.js web_root/history.min.js::gzip web_root/components.js::gzip web_root/main.css::gzip web_root/bundle.js::gzip web_root/index.html::gzip web_root/main.js::gzip certs/* > packed_fs.c
cc main.c mongoose.c net.c packed_fs.c -W -Wall -Wextra -Wundef -Wshadow -Wdouble-promotion -Wconversion -Wno-sign-conversion -fno-common -ffunction-sections -fdata-sections -g3 -Os -I. -DMG_ENABLE_PACKED_FS=1  -o ./example              
./example        
592bde1 3 mongoose.c:3996:mg_listen     1 4 http://0.0.0.0:8000
592bde1 3 mongoose.c:3996:mg_listen     2 5 https://0.0.0.0:8443
592bde1 3 mongoose.c:3971:mg_connect    3 -1 udp://time.google.com:123
592bde1 3 mongoose.c:3971:mg_connect    4 -1 udp://8.8.8.8:53
592bde1 3 mongoose.c:7891:mg_send       4 6 0:0:0 33 err 0
592bde8 3 mongoose.c:8058:read_conn     4 6 0:0:0 97 err 0
592bde8 3 mongoose.c:289:dns_cb         3 time.google.com is 216.239.35.0
592bde8 3 mongoose.c:7891:mg_send       3 7 0:0:0 48 err 0
592be04 3 mongoose.c:8058:read_conn     3 7 0:0:0 48 err 0
592be04 3 mongoose.c:7712:sntp_cb       3 got time: 1736961806006 ms from epoch
592be04 3 mongoose.c:3945:mg_close_conn 3 7 closed
```

This is missing in your invokation:
```
node pack.js web_root/history.min.js::gzip web_root/components.js::gzip web_root/main.css::gzip web_root/bundle.js::gzip web_root/index.html::gzip web_root/main.js::gzip certs/* > packed_fs.c
```

https://github.com/cesanta/mongoose/blob/d386b1c12dbf965911c292efc84e26fa33cdbfb0/reference-projects/windows-macos-linux/web-ui-dashboard/Makefile#L8
https://github.com/cesanta/mongoose/blob/d386b1c12dbf965911c292efc84e26fa33cdbfb0/reference-projects/windows-macos-linux/web-ui-dashboard/Makefile#L43-L45

You seem to have a bogus packed_fs.c

embetrix, 2025-01-15:
@scaprile thanks for the clarification,  the problem was using:
` make -C reference-projects/windows-macos-linux/web-ui-dashboard/`

it works if using:
```
cd reference-projects/windows-macos-linux/web-ui-dashboard/
make
```

closed 2025-01-15 by embetrix

scaprile, 2025-01-15:
-C should change directory and then make. Maybe your make tool has some attitude. Works for us here.