cesanta/mongoose #3030: Server access forbidden from esp32

dattasaurabh82, 2025-02-01:
# Things I have tried

## Setup and context

1. I'm using [Adafruit Feather Huzzah 32](https://www.adafruit.com/product/3405) board that uses [ESP-WROOM-32](https://www.mouser.de/new/espressif/espressif-esp32-wroom-32-modules/)
2. I'm using Arduino Development environment and Arduino IDE version `2.3.4`
3. I'm trying the basic `Single On/Off Toggle` example from `Files` from the wizard.
4. I just started exploring mongoose for esp32 from this morning.

<br>
<br>

> __Wizard__ is amazing 😘 but there are few hiccups I'm facing (I also noticed on discord other people are having some of these issues).

<br>

## Issues

1. Project Download and sync functionality is basic and expected, but not working correctly or as intended from UX perspective.
   1. After the __FIRST TIME__ source code generation, I can't download the project directly from the sync button.
   2. I have to instead first generate the source code in wizard.
   3. Then in the computer, create a desired arduino project directory with a desired project name. It creates a `[dir_name].ino` in `[dir_name]/` (path: `[dir_name]/[dir_name].ino`) create source an empty directory, upload that directory, generate the source code according to the target uC and then I can sync.
   4. Then come back to wizard, load the directory and then sync.
   5. At this point, the ino file name changes from `[dir_name].ino` to `wizard.ino`
   6. Why can't we
      1. Give the project in wizard a project name? That way the `wizard.ino` is replaced as `[project_name].ino` and so if the project downloadable directory?
      2. After src code generation, it could give us option to "sync" or "download once". If sync was selected, then in the download folder there's also a git tracker - to help with sync?

2. Source code generated, has some errors - Logging issues of DHCP IP addr. No string formatting
   1. In the current `wizard.ino`, there is a line: `MG_INFO((Connected; IP: %s", WiFi.localIP()));`. The parentheses are mismatched - there's an extra opening parenthesis. The correct one simple should be: `MG_INFO(("Connected; IP: %s", WiFi.localIP()));`
   2. `MG_INFO(("Connected; IP: %s", WiFi.localIP()));` doesn't print the DHCP IP address.

       ```c++
       // ...
       // MG_INFO(("Connected; IP: %s", WiFi.localIP())); prints empty IP or some single char gibberish
       WiFi.mode(WIFI_STA);
       WiFi.begin(WIFI_SSID, WIFI_PASS);
       
       while (WiFi.status() != WL_CONNECTED){
           delay(100);
       } 
       
       MG_INFO(("Connected; IP: %s", WiFi.localIP()));

       // Whereas using Serial.println(...), prints correctly...
       Serial.println("Connected to WiFi!");
       Serial.print("IP Address: ");
       Serial.println(WiFi.localIP());
       ```

       Serial Output:

       ```txt
       ...
       15:30:50.905 -> entry 0x400805b8
       15:30:51.998 -> 326    2 mongoose_ws_hello_world.ino:21 Connected; IP: �A?
       15:30:51.998 -> Connected to WiFi!
       15:30:51.998 -> IP Address: 192.168.1.117
       15:30:51.998 -> 327    2 mongoose_impl.c:685:mongoose_i Starting HTTP listener
       15:30:51.998 -> 334    3 mongoose.c:3996:mg_listen      1 48 http://0.0.0.0:80
       15:30:51.998 -> 335    2 mongoose_impl.c:689:mongoose_i Starting HTTPS listener
       15:30:51.998 -> 337    3 mongoose.c:3996:mg_listen      2 49 https://0.0.0.0:443
       15:30:52.032 -> 342    2 mongoose_impl.c:727:mongoose_i Mongoose init complete, calling user init
       15:30:52.032 -> 344    3 mongoose_glue.c:11:glue_init   Custom init done
       ...
       ```

       > As you can see the `MG_INFO` is not printing the IP_ADDR.

       The IPAddress object isn't a C-style string that can be printed with `%s`. The Arduino's Serial class knows how to properly convert and print an IPAddress object. However, `MG_INFO` (from `mongoose.h`) is trying to treat the IPAddress as a string pointer, which is not correct.
[... 136 lines cut ...]
        16:38:09.698 -> 5fc2   3 mongoose.c:8203:accept_conn    9 50 accepted 192.168.1.146:50300 -> 0.0.0.0:80
        16:38:09.698 -> 5fc8   3 mongoose.c:8057:read_conn      9 50 0:0:0 256 err 0
        16:38:09.698 -> 5fc9   1 mongoose.c:2389:http_cb        HTTP parse, 256 bytes
        16:38:09.698 -> 0000   16 03 01 06 e0 01 00 06 dc 03 03 e3 b2 9f d7 dd   ................
        16:38:09.730 -> 5fd6   3 mongoose.c:3945:mg_close_conn  9 50 closed
        16:38:09.730 -> 5fe9   3 mongoose.c:8203:accept_conn    10 50 accepted 192.168.1.146:50302 -> 0.0.0.0:80
        16:38:09.730 -> 5fec   3 mongoose.c:8057:read_conn      10 50 0:0:0 256 err 0
        16:38:09.730 -> 5fed   1 mongoose.c:2389:http_cb        HTTP parse, 256 bytes
        16:38:09.730 -> 0000   16 03 01 07 00 01 00 06 fc 03 03 dc 61 0f 71 b4   ............a.q.
        16:38:09.763 -> 5ffa   3 mongoose.c:3945:mg_close_conn  10 50 closed
        16:38:13.699 -> 6f5d   3 mongoose.c:8203:accept_conn    11 50 accepted 192.168.1.146:50344 -> 0.0.0.0:80
        16:38:13.699 -> 6f60   3 mongoose.c:8057:read_conn      11 50 0:0:0 256 err 0
        16:38:13.699 -> 6f61   3 mongoose.c:8057:read_conn      11 50 0:256:0 254 err 0
        16:38:13.699 -> 6f6d   3 mongoose_impl.c:470:http_ev_ha 11 GET / 0 -> 403
        16:38:13.752 -> 6f6f   3 mongoose.c:8067:write_conn     11 50 snd 65/256 rcv 0/512 n=65 err=0
        16:38:13.752 -> 6f8b   3 mongoose.c:8203:accept_conn    12 51 accepted 192.168.1.146:50349 -> 0.0.0.0:80
        16:38:18.844 -> 835c   3 mongoose.c:8057:read_conn      12 51 0:0:0 -1 err 128
        16:38:18.844 -> 835e   3 mongoose.c:3945:mg_close_conn  12 51 closed
        16:38:18.844 -> 835f   3 mongoose.c:8057:read_conn      11 50 0:0:0 -1 err 128
        16:38:18.844 -> 836b   3 mongoose.c:3945:mg_close_conn  11 50 closed
       ```

       I tried changing the port from `80` to something else in `mongoose_impl.c` and to also made sure to see the DHCP IP wasn't conflicting. No success.

       What am I doing wrong?

scaprile, 2025-02-01:
Please fill the issue template
Please report one issue per issue. Please make sure it is a Mongoose issue, if you are just asking a question, open a question instead. Please be specific, we won't check your code for errors
Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).

closed 2025-02-01 by scaprile

dattasaurabh82, 2025-02-02:
1. **Please fill the issue template**  -- Didn't see an issue template for this repo. 
2. **Please fill the issue template ...** -- Makes sense, sorry for mixing.   
3. **Please do not report Wizard issues in the Mongoose repo ...** -- Thanks for clarifying

scaprile, 2025-02-02:
> 1. **Please fill the issue template**  -- Didn't see an issue template for this repo.

Fixed,
Sorry about that, Github changed the schema and did not migrate our templated to their new schema