cesanta/mongoose #3031: RST ACK packet does not close connection server-side w/ OpenSSL

eoinf01, 2025-02-04:
- My goal is: To use mongoose as a web socket/http server and have the connection close and cleaned up server side in the event of RST ACK packet from the client
- My actions were: Upgrading from 7.12 to any version greater than that. I am building the library with the following flag: 
-DMG_TLS=MG_TLS_OPENSSL. I am not using the libraries built in implementation of the TCP/IP stack.

- My expectation was: In relation to the web socket server, for the forcibly disconnect message (RST,ACK) sent from the client to be handled the same as in previous version (7.12). 
- The result I saw:
```
39378e6a 4 mongoose.c:7761:mg_mgr_poll  3 r- Tchrc 0 0
39378e6a 4 mongoose.c:7309:recv_raw     3 -1 10054
39378e6a 3 mongoose.c:7356:read_conn    3 0x638 snd 302/2048 rcv 0/2048 n=-2 err=0
39378e6a 4 mongoose.c:7761:mg_mgr_poll  1 -- tchrc 0 0
39378e6a 4 mongoose.c:7761:mg_mgr_poll  3 rw Tchrc 0 0
39378e6a 4 mongoose.c:7309:recv_raw     3 -1 10054
39378e6a 3 mongoose.c:7356:read_conn    3 0x638 snd 302/2048 rcv 0/2048 n=-2 err=0
39378e6a 4 mongoose.c:7187:mg_io_send   3 -1 10054
39378e6a 3 mongoose.c:7367:write_conn   3 1592 snd 302/2048 rcv 0/2048 n=-2 err=10054
```
The forcibly disconnect message (RST,ACK) from the client results in the server failing to close the connection. write_conn passes MG_IO_WAIT not closing the connection. The event handler is not executed with the MG_EV_CLOSE event.

[client-pcap.zip](https://github.com/user-attachments/files/18658548/client-pcap.zip)
[server-pcap.zip](https://github.com/user-attachments/files/18658549/server-pcap.zip)

- My question is: 
- 
Is this expected or as a result of the added use of the BIO for OpenSSL? Previously, from my view what was closing connection server side was SSL_ERROR_SSL returned from SSL_get_error. Now, I only get SSL_ERROR_WANT_READ or SSL_ERROR_WANT_WRITE possibly because the BIO read/write handler is setting BIO_set_retry_write/BIO_set_retry_read?

# Environment

- mongoose version: >= 7.13
- Compiler/IDE and SDK: Microsoft Visual Studio 2019, MSCV 14.29.30133, Windows SDK 10.0.26100.0
- Target hardware/board: _?_
- Connectivity chip/module: _?_
- Target RTOS/OS (if applicable): Windows 10

scaprile, 2025-02-04:
7.13 is ages old, please upgrade to 7.16. We do not support old versions unless you are a paying customer, in which case please contact Support as per your contract.
We did have some incomplete transfers in past versions, maybe this is your problem.
We do not support OpenSSL on Windows, it is not standard and you have to compile it yourself, hence we don't know what version you are using nor your mods.
We do not handle the TCP/IP stack in Windows, it is Windows.

cpq, 2025-02-05:
@eoinf01 could you share a minimal example code we could use to reproduce the issue?

scaprile, 2025-02-05:
Wild guess, what happens if you change 
https://github.com/cesanta/mongoose/blob/ef8e7419fa704838f8513fd1c6e99363e5dcc182/src/sock.c#L291
to
```
if (c->rtls.len == 0 || m <= 0) {
```
? (m less than **or equal** to 0)

cpq, 2025-02-12:
@eoinf01 Eoin, could you follow up that please?

eoinf01, 2025-02-17:
@scaprile @cpq With the suggested change of adding m less than or equal to zero to the if statement, the same scenario occurs.

c->rtls.len equals zero constantly and mg_tls_pending also returns 0 constantly. Also, recv_raw returns MG_IO_RESET not MG_IO_ERROR so it doesn't fall into the block statement to check the length of rtls buffer or the value returned by mg_tls_recv.  As mg_tls_recv returns MG_IO_WAIT on each poll which is the value getting passed to iolog.

Server Example Code:
```
#include "mongoose.h"

static const char *s_listen_on = "ws://localhost:8000";
static const char *s_web_root = ".";

// This RESTful server implements the following endpoints:
//   /websocket - upgrade to Websocket, and implement websocket echo server
//   /rest - respond with JSON string {"result": 123}
//   any other URI serves static files from s_web_root
static void fn(struct mg_connection *c, int ev, void *ev_data) {
  if(ev == MG_EV_CLOSE){
    printf("Websocket connection closed.");
  }
  else if(ev == MG_EV_ACCEPT){
    struct mg_tls_opts tlsOpts;
    memset(&tlsOpts, 0, sizeof(tlsOpts));
    size_t size = 0; 
    // read CA certs from plain file
    tlsOpts.ca = mg_file_read(&mg_fs_posix,"C:\\xx\\ca.crt");
    tlsOpts.cert = mg_file_read(
        &mg_fs_posix,
       "C:\\xx\\client.crt");
    tlsOpts.key = mg_file_read(
        &mg_fs_posix,
        "C:\\xx\\client.key");
    mg_tls_init(c, &tlsOpts);

    free(tlsOpts.ca.buf);
    free(tlsOpts.cert.buf);
    free(tlsOpts.key.buf);
  }
  else if (ev == MG_EV_OPEN) {

    // c->is_hexdumping = 1;
  } else if (ev == MG_EV_HTTP_MSG) {
    struct mg_http_message *hm = (struct mg_http_message *) ev_data;
    if (mg_match(hm->uri, mg_str("/websocket"), NULL)) {
      // Upgrade to websocket. From now on, a connection is a full-duplex
      // Websocket connection, which will receive MG_EV_WS_MSG events.
      mg_ws_upgrade(c, hm, NULL);
    } else if (mg_match(hm->uri, mg_str("/rest"), NULL)) {
      // Serve REST response
      mg_http_reply(c, 200, "", "{\"result\": %d}\n", 123);
    } else {
      // Serve static files
      struct mg_http_serve_opts opts = {.root_dir = s_web_root};
      mg_http_serve_dir(c, ev_data, &opts);
    }
  } else if (ev == MG_EV_WS_MSG) {
    // Got websocket frame. Received data is wm->data. Echo it back!
    struct mg_ws_message *wm = (struct mg_ws_message *) ev_data;
    mg_ws_send(c, wm->data.buf, wm->data.len, WEBSOCKET_OP_TEXT);
  }
}

int main(void) {
  struct mg_mgr mgr;  // Event manager
  mg_mgr_init(&mgr);  // Initialise event manager
  mg_http_listen(&mgr, s_listen_on, fn, NULL);  // Create HTTP listener
  for (;;) mg_mgr_poll(&mgr, 1000);             // Infinite event loop
  mg_mgr_free(&mgr);
  return 0;
}
```
Client Example Code:
```
#include "mongoose.h"

static const char *s_url = "ws://localhost:8000/websocket";

// Print websocket response and signal that we're done
static void fn(struct mg_connection *c, int ev, void *ev_data) {
  if (ev == MG_EV_OPEN) {
  } else if (ev == MG_EV_CONNECT) {
    struct mg_tls_opts tlsOpts;
    memset(&tlsOpts, 0, sizeof(tlsOpts));
    size_t size = 0;  // read CA certs from plain file
    tlsOpts.ca = mg_file_read(&mg_fs_posix,"C:\\xx\\ca.crt");
    tlsOpts.cert = mg_file_read(
        &mg_fs_posix,
       "C:\\xx\\server.crt");
    tlsOpts.key = mg_file_read(
        &mg_fs_posix,
        "C:\\xx\\server.key");
    mg_tls_init(c, &tlsOpts);

    free(tlsOpts.ca.buf);
    free(tlsOpts.cert.buf);
    free(tlsOpts.key.buf);
  } else if (ev == MG_EV_ERROR) {
    // On error, log error message
    MG_ERROR(("%p %s", c->fd, (char *) ev_data));
  } else if (ev == MG_EV_WS_OPEN) {
    // When websocket handshake is successful, send message
    mg_ws_send(c, "hello", 5, WEBSOCKET_OP_TEXT);
  } else if (ev == MG_EV_WS_MSG) {
    // When we get echo response, print it
    struct mg_ws_message *wm = (struct mg_ws_message *) ev_data;
    printf("GOT ECHO REPLY: [%.*s]\n", (int) wm->data.len, wm->data.buf);
  }
}

int main(void) {
  struct mg_mgr mgr;        // Event manager
  bool done = false;        // Event handler flips it to true
  struct mg_connection *c;  // Client connection
  mg_mgr_init(&mgr);                     // Initialise event manager
  mg_log_set(MG_LL_DEBUG);                          // Set log level
  c = mg_ws_connect(&mgr, s_url, fn, &done, NULL);  // Create client
  while (c) mg_mgr_poll(&mgr, 1000);                // Wait for echo
  mg_mgr_free(&mgr);                                // Deallocate resources
  return 0;
}
```

For the client code it doesn't forcibly reset the connection programmatically, however if you kill the process forcibly from the console, the RST ACK packet will be sent and the mongoose server won't clean up the connection.

scaprile, 2025-02-17:
OK, thank you, that was just a wild guess on a hunch. I'll be studying this as soon as possible.

scaprile, 2025-02-17:
Works for me in Linux
Can't reproduce in Windows, don't have OpenSSL. Works for me with HEAD (7.17 on Wednesday) with our built-in TLS
I see Windows returns -3 (MG_IO_RESET) and errno = 10054 when resetting; however, connection gets closed, so either this is related to OpenSSL or it only happens after feeding some amount of data.

I suggest you change
https://github.com/cesanta/mongoose/blob/49ce25511e204a0e8034c2f1b387f7fd3b821308/src/sock.c#L290
to
```
      if (n == MG_IO_ERR || n == MG_IO_RESET) {
```
and let us know if that solves your issue.

scaprile, 2025-02-18:
Fixed in #3042, merged in HEAD, will be included in 7.17, due tomorrow
Closing, let us know if it needs to be reopened

closed 2025-02-18 by scaprile