cesanta/mongoose #3060: mg_json_get_long returns zero for an invalid string

jhsaren, 2025-03-01:
# Description and example code

I'm writing a simple http interface for an embedded device and trying to validate the POST json. I used a simple mg_json_get_long to read the value of a parameter. Since the function does not report on errors, I used LONG_MIN to see if something goes wrong or the parameter is not given in the json. I found that a string "f42" gives me the default value LONG_MIN but the string "-f42" gives zero. I thing this is a bug. If this is not bug, how to check the validity of the json value to be an integer? Simple code which generates the error:
```
#include "mongoose/mongoose.h"
#include <limits.h>
#include <stdio.h>

int main() {
    struct mg_str s1 = mg_str( "42" );
    struct mg_str s2 = mg_str( "-42" );
    struct mg_str s3 = mg_str( "f42" );
    struct mg_str s4 = mg_str( "-f42" );

    long val1 = mg_json_get_long( s1, "$", LONG_MIN );
    long val2 = mg_json_get_long( s2, "$", LONG_MIN );
    long val3 = mg_json_get_long( s3, "$", LONG_MIN );
    long val4 = mg_json_get_long( s4, "$", LONG_MIN );

    printf( "val1 = %ld\n", val1 );
    printf( "val2 = %ld\n", val2 );
    printf( "val3 = %ld\n", val3 );
    printf( "val4 = %ld\n", val4 );
    
    return 0;
}
```
The output of the program:
```
val1 = 42
val2 = -42
val3 = -9223372036854775808
val4 = 0
```

My mongoose version is 7.16 and the compiler I used is gcc 11.4.0. I found the issue in a linux (ubuntu 22.04).

Best regards,
Jan

scaprile, 2025-03-01:
Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).

https://mongoose.ws/documentation/#mg_json_get_long

### mg\_json\_get\_long()

```c
long mg_json_get_long(struct mg_str json, const char *path, long default_val);
```

Fetch integer numeric (long) value from the json string `json` at JSON path
`path`. Return it if found, or `default_val` if not found.

Parameters:
- `json` - a string containing valid JSON
- `path` - a JSON path. Must start with `$`
- `default_val` - a default value for the failure case

Return value: found value, or `default_val` value

Usage example:

```c
long a = mg_json_get_long(mg_str("[123]", "$a", -1));   // a = -1
long b = mg_json_get_long(mg_str("[123]", "$[0]", -1)); // b = 123
```

closed 2025-03-01 by scaprile

scaprile, 2025-03-01:
The point here is that the alleged bug is not due to a parsing error but an invalid json string. Your JSON string is not a valid JSON string and you pretend that the parser was able to craft magic out of it.

Let's read the doc again:

> Parameters:
> 
> - json - a string containing valid JSON

- json - a string containing _**valid** JSON_

jhsaren, 2025-03-01:
Thank you for the fast reply. Yes, the invalid string "-foo42" is not valid JSON, but this is also true for "foo42". I shall find an easy way to first check the validity of the json. Is it possible in mongoose? Thank you for the help. 
Best regards, Jan