cesanta/mongoose #3079: Unwanted websocket connection interruption introduced with 7028417

linkjumper, 2025-03-21:
The websocket connection is now closed immediatly. A `&& is_websocket` check might be missing here: https://github.com/cesanta/mongoose/blob/master/mongoose.c#L2469

## Environment
- Error is introduced with: 7028417d88089a2c9d9c28f210ccd3484de2769c
- mongoose version: 7.17
- Target hardware/board: embededd linux

scaprile, 2025-03-21:
Thank you. Before we blindly do what you tell us, please honor the issue template and properly describe your scenario, what you do, what you see, and why it is not working for you.
Our code is built and tested on every change and every night on a CI/CD pipeline, though there might be some path that escapes our tests.
Please sniff your network (using Wireshark, for example) and observe the traffic.
If, after your network inspection, you find an actual issue, fill the issue template, and provide us with pertinent information (including useful network capture files; no snapshots, please attach capture files) in order to reproduce and fix the issue. We don't inspect your code for errors, please send the minimum piece of code that can reproduce the issue
Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).

This issue will be reopened once and if you provide enough information for us to reproduce and fix what you claim to be an issue.
Thank you

scaprile, 2025-03-21:
> The websocket connection is now closed immediatly

No, it is not
https://mongoose.ws/documentation/tutorials/websocket/websocket-server/
https://mongoose.ws/documentation/tutorials/websocket/websocket-client/
```
$ make
cc main.c mongoose.c        -W -Wall -Wextra -g -I.   -DMG_ENABLE_LINES -DMG_TLS=MG_TLS_BUILTIN -o example                                    
./example                    
Starting WS listener on ws://localhost:8000/websocket
```
```
$ make
cc main.c mongoose.c        -W -Wall -Wextra -g -I.    -DMG_TLS=MG_TLS_BUILTIN -o example                                    
./example                    
11b78a57 3 mongoose.c:3972:mg_connect   1 -1 ws://localhost:8000/websocket
11b78a57 3 mongoose.c:8499:mg_connect_r 1 4 -> 127.0.0.1:8000 pend
11b78a57 3 mongoose.c:8419:write_conn   1 4 snd 157/16384 rcv 0/0 n=157 err=0
11b78a57 2 mongoose.c:8203:iolog        
-- 1 127.0.0.1:36748 -> 127.0.0.1:8000 157
0000   47 45 54 20 2f 77 65 62 73 6f 63 6b 65 74 20 48   GET /websocket H
0010   54 54 50 2f 31 2e 31 0d 0a 55 70 67 72 61 64 65   TTP/1.1..Upgrade
0020   3a 20 77 65 62 73 6f 63 6b 65 74 0d 0a 48 6f 73   : websocket..Hos
0030   74 3a 20 6c 6f 63 61 6c 68 6f 73 74 0d 0a 43 6f   t: localhost..Co
0040   6e 6e 65 63 74 69 6f 6e 3a 20 55 70 67 72 61 64   nnection: Upgrad
0050   65 0d 0a 53 65 63 2d 57 65 62 53 6f 63 6b 65 74   e..Sec-WebSocket
0060   2d 56 65 72 73 69 6f 6e 3a 20 31 33 0d 0a 53 65   -Version: 13..Se
0070   63 2d 57 65 62 53 6f 63 6b 65 74 2d 4b 65 79 3a   c-WebSocket-Key:
0080   20 58 71 50 7a 38 46 4a 74 31 2f 4c 44 76 30 7a    XqPz8FJt1/LDv0z
0090   6d 70 41 49 7a 79 51 3d 3d 0d 0a 0d 0a            mpAIzyQ==....   
11b78a57 3 mongoose.c:8408:read_conn    1 4 0:0:0 129 err 0
11b78a58 2 mongoose.c:8203:iolog        
-- 1 127.0.0.1:36748 <- 127.0.0.1:8000 129
0000   48 54 54 50 2f 31 2e 31 20 31 30 31 20 53 77 69   HTTP/1.1 101 Swi
0010   74 63 68 69 6e 67 20 50 72 6f 74 6f 63 6f 6c 73   tching Protocols
0020   0d 0a 55 70 67 72 61 64 65 3a 20 77 65 62 73 6f   ..Upgrade: webso
0030   63 6b 65 74 0d 0a 43 6f 6e 6e 65 63 74 69 6f 6e   cket..Connection
0040   3a 20 55 70 67 72 61 64 65 0d 0a 53 65 63 2d 57   : Upgrade..Sec-W
0050   65 62 53 6f 63 6b 65 74 2d 41 63 63 65 70 74 3a   ebSocket-Accept:
0060   20 38 78 49 67 67 6a 75 63 63 35 70 4a 4f 64 44    8xIggjucc5pJOdD
0070   77 68 4c 4c 6a 61 38 72 71 38 5a 30 3d 0d 0a 0d   whLLja8rq8Z0=...
0080   0a                                                .               
11b78a58 3 mongoose.c:8419:write_conn   1 4 snd 11/16384 rcv 0/16384 n=11 err=0
11b78a58 2 mongoose.c:8203:iolog        
-- 1 127.0.0.1:36748 -> 127.0.0.1:8000 11
0000   81 85 02 90 3b ac 6a f5 57 c0 6d                  ....;.j.W.m     
11b78a58 3 mongoose.c:8408:read_conn    1 4 0:0:0 7 err 0
11b78a58 2 mongoose.c:8203:iolog        
-- 1 127.0.0.1:36748 <- 127.0.0.1:8000 7
0000   81 05 68 65 6c 6c 6f                              ..hello         
GOT ECHO REPLY: [hello]
11b78a59 3 mongoose.c:3946:mg_close_con 1 4 closed
11b78a59 3 mongoose.c:4044:mg_mgr_free  All connections closed
```

closed 2025-03-21 by scaprile