cesanta/mongoose #3082: mg_connection has empty mg_addr remote ip

PripyatAutomations, 2025-03-23:
- My goal is: Get IP of the connected client
- My actions were:  Attmpted in my code as well as a simple example program below
- My expectation was:  To see the IP of the client printed
- The result I saw:
 HTTP server started on http://0.0.0.0:8000
Client connected: IP = 0.0.0.0, Port = 53804

- My question is:  Is this broken or am i missing something?
- 
# Environment

- mongoose version: latest git
- Compiler/IDE and SDK:  gcc on debian 12
- Target hardware/board:  x86/64 generic
- Connectivity chip/module: n/a
- Target RTOS/OS (if applicable): debian 12
- 

# Code
#include "mongoose.h"
#include <arpa/inet.h>
#include <stdio.h>

void event_handler(struct mg_connection *c, int ev, void *ev_data) {
    if (ev == MG_EV_ACCEPT) {
        char ip[INET6_ADDRSTRLEN];  // Buffer for the IP string

        // Check if the connection is IPv6 or IPv4
        if (c->rem.is_ip6) {
            inet_ntop(AF_INET6, c->rem.ip, ip, sizeof(ip));
        } else {
            inet_ntop(AF_INET, &c->rem.ip[12], ip, sizeof(ip));  // IPv4 is stored in the last 4 bytes
        }

        printf("Client connected: IP = %s, Port = %d\n", ip, ntohs(c->rem.port));
    }

    if (ev == MG_EV_HTTP_MSG) {
        struct mg_http_message *hm = (struct mg_http_message *) ev_data;
        mg_http_reply(c, 200, "", "Hello from Mongoose!\n");
    }
}

int main(void) {
    struct mg_mgr mgr;
    mg_mgr_init(&mgr);

    // Start HTTP server (use https://<addr>:<port> for TLS/SSL)
    if (mg_http_listen(&mgr, "http://0.0.0.0:8000", event_handler, NULL) == NULL) {
        printf("Error starting HTTP server\n");
        return 1;
    }

    printf("HTTP server started on http://0.0.0.0:8000\n");

    // Run the event manager
    for (;;) {
        mg_mgr_poll(&mgr, 1000);  // Poll every 1 second
    }

    mg_mgr_free(&mgr);  // Clean up after exit
    return 0;
}

closed 2025-03-24 by scaprile

PripyatAutomations, 2025-03-24:
Apologies for not explaining my issue more clearly.

 In my program, i need to access the IP of the remote end of the http session, so i can determine if the user is allowed to perform the requested action.

     MG_INFO(("%M", mg_print_ip_port, &c->rem)); indeed displays it, but i need to compare it to my ACL list. Sorry if i missed this while reviewing the documentation. Thanks for an excellent library!

scaprile, 2025-03-24:
Well, `mg_print_ip_port` doesn't do other than access the fields in `c->rem`, so. evidently, if it prints your IP then mg_connection does have mg_addr remote ip and is not empty, hence your subject does not hold true,
Evidently, whatever you do, you are not doing it the right way.
I've already given you pointers to the struct docs, here is how that function accesses the struct https://github.com/cesanta/mongoose/blob/c84826e84e254f5acd26044bec923e67f41faf43/src/printf.c#L81-L111, and Mongoose already has ACL management functions:
https://mongoose.ws/documentation/#mg_check_ip_acl
https://github.com/cesanta/mongoose/blob/c84826e84e254f5acd26044bec923e67f41faf43/src/util.c#L109-L125
though it doesn't handle IPv6 addresses. Nevertheless, you can also see there how to access the struct fields.