cesanta/mongoose #3085: Has no api for change min/max tls version

AaronKonishi, 2025-03-27:
- My goal is: limit tls version
- My actions were: limit tls version
- The result I saw: in mongoose.h has struct mg_tls
- My question is: How to set min/max tls version？

# Environment

- mongoose version: 7.17
- Compiler/IDE and SDK:  cmake, use MG_TLS_MBED
- Target hardware/board: H3
- Target RTOS/OS (if applicable): Linux

scaprile, 2025-03-27:
No, it does not.
Use MbedTLS calls
Thank you

AaronKonishi, 2025-03-27:
Hello, for code similar to following, should I call the mbedtls interafce before or after mg_tls_init(c, &m_opts); Can you provide details on which specific interface to call, eg?
```
void cb(struct mg_connection *c, int ev, void *ev_data) {
    switch(ev) {
        case MG_EV_ACCEPT:
            mg_tls_init(c, &m_opts);
            break;
        ......
}
......
```

scaprile, 2025-03-27:
That depends on MbedTLS and where it wants you to do what you want to do with it.
If it is before: before, if it is after: after, if it is another init function or different paramenters, you'll need to change source code.

Wrt Mongoose:
Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).
https://mongoose.ws/documentation/tutorials/tls/

https://github.com/cesanta/mongoose/blob/master/src/tls_mbed.c

https://github.com/cesanta/mongoose/blob/c84826e84e254f5acd26044bec923e67f41faf43/src/tls_mbed.c#L97-L151

If you are a paying customer or have a compelling use case, please contact Support. Thank you.