cesanta/mongoose #3099: TLS “opts”, does not accept RSA Private Key ??

paloky, 2025-04-11:
- My goal is:  Connect to to AWS MQTT IoT Core
- My actions were:  Set a RSA Private Key in opts.key
- The result I saw:   "1 mongoose.c:432:mg_error        2 -1 Expected EC PRIVATE KEY or PRIVATE KEY<CR><LF>"
- My question is:  How use a RSA private Key

# Environment
- mongoose version:  7.17
- Compiler/IDE and SDK:   STM32 Cube IDE 1.18
- Target hardware/board:  Custom board with STM32F402
- Connectivity chip/module:  ETH+MQTT
- Target RTOS/OS (if applicable):  ---

#
Hi.
I try to config the TLS options to connect with AWS IotCore over MQTTS.
The problem seems to be that the library does not support “RSA” and only supports “EC”.  Is this the case?

I have followed the code and I see that it fails to verify the certificate header, because it is expecting to find “----BEGIN EC PRIVATE KEY----”.
but found a "-----BEGIN RSA PRIVATE KEY-----"

How can I use a private key using RSA ?

Than's a lot.

cpq, 2025-04-11:
@paloky you should generate EC instead. Mongoose's TLS implemetation is EC.
See https://www.youtube.com/watch?v=03nlRQpZrM4
Let me know how it goes.

paloky, 2025-04-14:
Hi @cpq 

I have managed to configure the certificates correctly in AWS.
It only works if you first create the certificate and then assign it to an Object (If you do it the other way around, first create the object and create the certificate from the object creation, the certificate is not activated).

Now I have another problem.
After 14 publications, I get a “HardFault” because of a “calloc”.  I put the “mg_tls_free()” inside the "MG_EV_CLOSE" event to free the memory of TLS, but it doesn't work.

I don't see where else memory is saved.
Do I have to free some other memory?

This is my handle code:

```

static void timer_fn(void *arg)
{
	// It's ready and have IP ?
	if (mgr.ifp->state == MG_TCPIP_STATE_READY)
	{
		struct mg_mgr *mgr = (struct mg_mgr *) arg;
		if (s_mqtt_conn == NULL)
		{
			struct mg_mqtt_opts opts = {.clean = true, .client_id=mg_str("100998_TEST_2")};
			s_mqtt_conn = mg_mqtt_connect(mgr, s_mqtt_url, &opts, ev_handler, NULL);
		}
	}
}

static void ev_handler(struct mg_connection *c, int ev, void *ev_data)
     {

	  if (ev == MG_EV_ERROR)
	  {
		  MG_INFO(("%lu ERROR:  %s ...", c->id, (char*)ev_data));
	  }

	  else if (ev == MG_EV_OPEN)
	  {
		MG_INFO(("%lu CONECTANDO CON BROKER:  %s ...", c->id, s_mqtt_url));
	  }

	  else if (ev == MG_EV_CONNECT)
	  {
		  MG_INFO(("%lu CONEXION ESTABLECIDA:  %s ", c->id, s_mqtt_url));

		  // Iniciando TLS
		  MG_INFO(("%lu CONFIGURANDO TLS.... ", c->id));
		  struct mg_tls_opts opts;
		  opts.ca = mg_str(s_tls_ca);
		  opts.cert = mg_str(s_tls_cert);
		  opts.key = mg_str(s_tls_key);
		  opts.name = mg_url_host(s_mqtt_url);

		  mg_tls_init(c, &opts);

	  }

	  else if (ev == MG_EV_TLS_HS)
	  {
		  MG_INFO(("%lu TLS handshake succeeded", c->id));
	  }

	  else if (ev == MG_EV_MQTT_OPEN)
	  {
		// Publicamos a la que se abre la conexión
		char response[100];
		struct mg_mqtt_opts opts_2 = {.qos = s_qos, .topic = mg_str("100998_TEST_2/test")};
		mg_snprintf(response, sizeof(response), "Soy un bSensys !!!");
		opts_2.message = mg_str(response);
		MG_INFO(("%lu PUBLICANDO EN EL TOPIC [ %s ]:  %s ...", c->id,  opts_2.topic.buf,  response));
		mg_mqtt_pub(c, &opts_2);

		// Guardamos el ID, para saber si se ha enviado
		pub_id = c->id;
	  }

	  else if (ev == MG_EV_MQTT_CMD)
	  {
		  struct mg_mqtt_message *respuesta = (struct mg_mqtt_message*) ev_data;
		  if (respuesta->cmd == MQTT_CMD_PUBACK)
		  {
			  MG_INFO(("%lu Recibido el PUB_ACK ....", c->id));
			  MG_INFO(("%lu CERRANDO CONEXION MQTT ....", c->id));
			  mg_mqtt_disconnect(c, NULL);
		  }
	  }

	  else if (ev == MG_EV_MQTT_MSG)
	  {
		 struct mg_mqtt_message *mm = (struct mg_mqtt_message *) ev_data;
		 MG_INFO(("%lu DATOS RECIBIDOS:  %.*s : %.*s", c->id, (int) mm->topic.len, mm->topic.buf,  (int) mm->data.len, mm->data.buf));

	  }

	  else if (ev == MG_EV_CLOSE)
	  {
	    MG_INFO(("%u CONEXION CERRADA", c->id));
	    s_mqtt_conn = NULL;
	    *(bool *) c->fn_data = true;
	    mg_tls_free(c);
	  }
	}

```

Thank's a lot.
Sergi

scaprile, 2025-04-14:
You are interfering with the normal MQTT handler by catching `MG_EV_MQTT_CMD` and breaking the flow by issuing a disconnect call there. See the examples, we *do not* do that. (1)

```
static void timer_fn(void *arg)
{
	// It's ready and have IP ?
	if (mgr.ifp->state == MG_TCPIP_STATE_READY)
```
Please don't do that, there is an event handler for the tcp/ip interface
https://mongoose.ws/documentation/#drivers-for-mongoose-built-in-tcpip-stack
https://github.com/cesanta/mongoose/blob/2557ab2e60b986a735413f2fdf1f01f6495cf39e/tutorials/tcpip/pcap-driver/main.c#L172
https://github.com/cesanta/mongoose/blob/2557ab2e60b986a735413f2fdf1f01f6495cf39e/tutorials/tcpip/tap-driver/main.c#L47

Please see our [documentation](https://mongoose.ws/documentation/),  and follow the guidelines in our [tutorials](https://mongoose.ws/documentation/#tutorials-and-examples).
https://mongoose.ws/documentation/tutorials/mqtt/mqtt-client-aws-iot/
https://mongoose.ws/documentation/tutorials/mqtt/mqtt-client/
https://mongoose.ws/documentation/tutorials/tls/#linux

You don't have to free memory you don't alloc, unless one of the Mongoose functions you use tells you to do so in its docs https://mongoose.ws/documentation/#api-reference , and I don't see any of them being called. If you check the examples linked above, we _don't_ do that, we _never_ call that function explicitly in a user handler.

(1): You are interfering with the normal MQTT handler by catching `MG_EV_MQTT_CMD` and breaking the flow by issuing a disconnect call there. The internal protocol handler is doing its job, if you want to see what is going on, you can, but please don't break the flow. See the examples, we *do not* do that.
Here's yet another expected MQTT flow: https://github.com/cesanta/mongoose/blob/2557ab2e60b986a735413f2fdf1f01f6495cf39e/test/unit_test.c#L572-L663

paloky, 2025-04-14:
Hi @scaprile 
I understand.

But then I do not know how to disconnect from the broker.
In all the examples of documentation, there is not 1 single example of disconnection. (Or at least I've seen it).

I have tried to put a flag in the Event Handler and perform the disconnection from the main “for” where the “mg_mgr_poll()” is doing, but this doesn't work either.

Do you have an example of how to call the “mg_mqtt_disconnect()” without breaking the flow?

Thank's a lot.

closed 2025-04-14 by paloky